arduino/Arduino #8274: Arduino 1.8.8 crashes on startup - Mac OSX 10.11.6 (El C.)

justb4, 2018-12-08:
Upgraded from 1.6.6. 

v1.8.8 immediately crashes on startup with exception in log console and crash report:

```
Process:               Arduino [1153]
Path:                  /Applications/development/Arduino.app/Contents/MacOS/Arduino
Identifier:            cc.arduino.Arduino
Version:               1.8.8 (1.8.8)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [1153]
User ID:               501

Date/Time:             2018-12-08 18:02:00.194 +0100
OS Version:            Mac OS X 10.11.6 (15G22010)
Report Version:        11
Anonymous UUID:        D68992EA-87DB-CDED-6913-AA92AAE020A6

Time Awake Since Boot: 740 seconds

System Integrity Protection: enabled

Crashed Thread:        0  AppKit Thread  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplicationAWT setAutomaticCustomizeTouchBarMenuItemEnabled:]: unrecognized selector sent to instance 0x7fd003d71ae0'
terminating with uncaught exception of type NSException
abort() called

```
Java version is:

```
java version "1.8.0_102"
Java(TM) SE Runtime Environment (build 1.8.0_102-b14)
Java HotSpot(TM) 64-Bit Server VM (build 25.102-b14, mixed mode)
```
Also removing `~/Library/Arduino15` did not help. 

Downloaded 1.8.6. : no problems, only slow File menu.

facchinm, 2018-12-10:
@everyone may you report if the bug is happening on all El Capitan (and eariler) macOS versions? Since the error refer to `setAutomaticCustomizeTouchBarMenuItemEnabled` it is surely related with https://github.com/arduino/Arduino/pull/8156 , but before reverting it would be nice to know when and why the bug is triggered.

cigarete, 2018-12-10:
I'm having the same issue. It opens then immediately crashes. Mid 2009 mac just upgraded to El Captain 10.11.6

Process:               Arduino [483]
Path:                  /Users/USER/Desktop/Arduino-2.app/Contents/MacOS/Arduino
Identifier:            cc.arduino.Arduino
Version:               1.8.8 (1.8.8)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [483]
User ID:               501

Date/Time:             2018-12-10 09:18:56.061 -0500
OS Version:            Mac OS X 10.11.6 (15G22010)
Report Version:        11
Anonymous UUID:        976FC489-6C4A-3509-EE06-9CB4705DE9C3

Time Awake Since Boot: 720 seconds

System Integrity Protection: enabled

Crashed Thread:        0  AppKit Thread  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplicationAWT setAutomaticCustomizeTouchBarMenuItemEnabled:]: unrecognized selector sent to instance 0x7fca71c5aab0'
terminating with uncaught exception of type NSException
abort() called

Model: MacBookPro5,3, BootROM MBP53.00AC.B03, 2 processors, Intel Core 2 Duo, 2.66 GHz, 4 GB, SMC 1.48f2
Graphics: NVIDIA GeForce 9400M, NVIDIA GeForce 9400M, PCI, 256 MB
Graphics: NVIDIA GeForce 9600M GT, NVIDIA GeForce 9600M GT, PCIe, 256 MB
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1067 MHz, 0x802C, 0x31364A53463235363634485A2D3147314631
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1067 MHz, 0x802C, 0x31364A53463235363634485A2D3147314631
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x8D), Broadcom BCM43xx 1.0 (5.106.98.102.30)
Bluetooth: Version 4.4.6f4, 3 services, 19 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: SSD2SC240G1SA754D117-459, 240.06 GB
Serial ATA Device: HL-DT-ST DVDRW  GS23N
USB Device: USB 2.0 Bus
USB Device: Card Reader
USB Device: USB 2.0 Bus
USB Device: Built-in iSight
USB Device: Composite Device
USB Device: USB Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: IR Receiver
USB Device: USB Bus
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus:

sandeepmistry, 2018-12-10:
Hi @justb4 and @cigarete,

Unfortunately we don't have an 10.11 machines around, so IDE 1.8.8 was only tested on 10.12, 10.13, and 10.14.

Could you please try the IDE version for this pull request to see if the change fixes this issue and provide your feedback. Thank you.

https://github.com/arduino/Arduino/pull/8281#issuecomment-445854643

justb4, 2018-12-10:
Hi @sandeepmistry thanks for quick action! Unfortunately the version in PR #8281 still crashes on my 10.11.6: 
```
Process:               Arduino [36064]
Path:                  /Users/USER/Desktop/Arduino.app/Contents/MacOS/Arduino
Identifier:            cc.arduino.Arduino
Version:               PR-8281-BUILD-814 (1.8.9)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [36064]
User ID:               501

Date/Time:             2018-12-10 17:05:13.186 +0100
OS Version:            Mac OS X 10.11.6 (15G22010)
Report Version:        11
Anonymous UUID:        D68992EA-87DB-CDED-6913-AA92AAE020A6

Sleep/Wake UUID:       7DC44E7B-054C-4F1D-AC7A-22CDCE0A0207

Time Awake Since Boot: 53000 seconds
Time Since Wake:       20000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  AppKit Thread  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplicationAWT setAutomaticCustomizeTouchBarMenuItemEnabled:]: unrecognized selector sent to instance 0x7fa792e34110'
terminating with uncaught exception of type NSException
abort() called

```

Looking into [your commit](https://github.com/arduino/Arduino/pull/8281/commits/feccea0b78c07109920c295ce57144995bb75811): could it be that the version-comparison still succeeds? On my system  `System.getProperty("os.version")` gives `10.11.6`. Rendering `VersionComparator.greaterThan("10.11.6", "10.11")) `, IMHO will return 1. 
Possibly you need to chop-off the minor `.6` part.

sandeepmistry, 2018-12-10:
@justb4 thanks for trying it!

Good catch, I've made another change to have a >= 10.12 check instead.

At your convenience please try the build from: https://github.com/arduino/Arduino/pull/8281#issuecomment-445904330

justb4, 2018-12-10:
@sandeepmistry yes this version above works for me, no crashing! Thanks!

sandeepmistry, 2018-12-10:
@justb4 thanks for the feedback!

sandeepmistry, 2018-12-11:
Closed via https://github.com/arduino/Arduino/pull/8281.

closed 2018-12-11 by sandeepmistry

cooneypc4, 2018-12-14:
@sandeepmistry I am having the same problem with El Capitan 10.11.4. When I try the link for the Mac zip file in #8281 , I get a 404: Page Not Found error. Any suggestions?

sandeepmistry, 2018-12-14:
@cooneypc4 please try the hourly build from the downloads page.

Please note, the change made to fix this issue was only for Mac OS X 10.11 and older.

cooneypc4, 2018-12-14:
@sandeepmistry Thanks for your help! It works now!

selpizza, 2018-12-15:
Hi
@sandeepmistry , I am having the same problem with Maverick 10.9.5. 
When I try the link for the Mac zip file in #8281 , I get "404 Not Found nginx".
Any suggestions?

per1234, 2018-12-15:
@selpizza the fix is now available in the hourly build of the Arduino IDE. That's why the test build in sandeepmistry's pull request is no longer available. So you only need to download and install the hourly build from the official Arduino IDE downloads page:
https://www.arduino.cc/en/Main/Software#hourly

selpizza, 2018-12-15:
Perfect!
Thank you very much @per1234 !

dirkblaze, 2019-01-15:
Hey!  Getting this crash with the OS X version I downloaded today:

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplicationAWT setAutomaticCustomizeTouchBarMenuItemEnabled:]: unrecognized selector sent to instance 0x7fc76b742fb0'
terminating with uncaught exception of type NSException
abort() called

facchinm, 2019-01-16:
@dirkblaze please use IDE version 1.8.7 or hourly until we don't release 1.8.9, thanks.

AhmadMuzaffar, 2019-02-22:
I faced the same problem on mac os 10.11.6

facchinm, 2019-02-22:
@AhmadMuzaffar please use IDE version 1.8.7 or hourly until we don't release 1.8.9, thanks.

sarahthebus, 2019-03-07:
All above versions of IDE crashing on opening, macOS Mojave 10.14.3

Please help

justinxreese, 2019-05-09:
This still happens in IDE 1.8.9 on macOS 10.12.1 on a 2016 MacBook that does not have a touchbar. #8281 only mentions checking the OS version but not if the computer has a touchbar. Is that possible and needed?

facchinm, 2019-05-13:
@sandeepmistry shouldn't macOS handle this internally starting from 10.12?

mkarliner, 2019-09-23:
Sigh. Just downloaded Arduino 1.8.10, Sierra 10.12 (on ahem, generic hardware). 
Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplicationAWT setAutomaticCustomizeTouchBarMenuItemEnabled:]: unrecognized selector sent to instance 0x6180001495e0'
terminating with uncaught exception of type NSException
abort() called

No problems with any other applications that I'm aware of.