arduino/Arduino #8276: aduino IDE didn`t send programs in ryzen cpu based computers

JasonCbrody, 2018-12-09:
My computer system is ryzen 2400g and b350m-HDV, and I`m using Arduino Uno.Arduino IDE is 1.8.8 Windows store version.
I tried to upload new program and test it out, but it spit errors and failed.
I tried all usb ports from the front, but it didn`t work.
Do I did something wrong? please help.

cmaglie, 2018-12-09:
Could you copy & paste the errors here?

cmaglie, 2018-12-09:
Also, may you also explain why you think the error it's related to ryzen CPU?

JasonCbrody, 2018-12-10:
avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x6e
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x6e
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x6e
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6e
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x6e
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x6e
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x6e
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x6e
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x6e
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x6e

avrdude done.  Thank you.

보드에 업로딩중에 문제 발생. 다음을  참고하세요.
(error occured while board uploading)
http://www.arduino.cc/en/Guide/Troubleshooting#upload
cause there is no issue when I uploaded the code with my intel laptop

per1234, 2020-10-05:
Since this problem is normally caused by a problem with the Arduino board and we didn't get any information that would imply it was caused by Arduino's software, I'll close this as invalid.

For others who find this issue when searching for a solution to this type of error message, please post on the Arduino Forum:
https://forum.arduino.cc/index.php?board=119.0
I'm sure we'll be able to help you out over there.

closed 2020-10-05 by per1234

sercona, 2022-04-26:
this is still a real problem, guys.

I'm using linux on a Ryzen 5 5600G system with a recent kernel 5.16.0 and arduino ide 1.8.12.

when I use the same exact environment on an older intel i3 system, I never get timeouts and flashing issues with avrdude.  but most of the time, the ryzen system wont flash the nano 328 chip.  I've also seen issues using usb to flash the wemos d1 (esp8266) chip over usb.  when I flash over wifi for that chip, it always works no matter which system (intel or amd).

for now, I've resorted to flashing on my intel box and building on my faster amd box, but we should find out why the usb timing is 'different' on amd.  this is a show-stopper bug since you can't reliably flash on modern ryzen's, with confidence.

(been using arduino for over 7 years, so I'm not a newbie, fwiw)

sercona, 2022-04-26:
also, while I'm reporting this from an amd ryzen box, my guess is its the usb chipset and not the cpu itself.  or something about a scheduler for sending out usb packets.  shame that the bootloader is so timing sensitive, but something about the motherboard chipsets (my guess) on amd is causing the delivery of usb packets to be out of tolerance for both old and new 328 nano bootloaders.  have not tested on other 328's but I use nano a lot quite a lot.  ch340 chip on my nano clones.

per1234, 2022-04-26:
Hi @sercona. Please check whether the problem still occurs when using the latest version of AVRDUDE directly:

https://github.com/avrdudes/avrdude/actions/runs/2145022135#artifacts

sercona, 2022-04-29:
good news, latest avrdude works fine on my ryzen box.

using version 6.99 that I configured and built from latest git source.

did a sudo make install and used that version as the binary and used the arduino conf file, as:

avrdude -C/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega328p -carduino -P/dev/ttyUSB0 -b57600 -D -Uflash:w:foo_test.ino.hex:i

and that worked great!  nothing else changed.  before I ran the make install, a pre-built .hex did not flash on my amd box, but flashed fine on my i3.  I grabbed the verbose output of the arduino ide build and used that as the command for manually testing avrdude.

on the most recent arduino ide, they use a very old version for some reason,  6.3-20190619 is the string.  2019?  isn't this 2022?  (grin)

so let this be a good enough reason to at least look at pulling in the latest avrude and pkging it with arduino ide.

wish I tried this earlier.  doh!  I'll also try it on my troublesome ESP chips that also fail under usb flash but work with OTA flash.

thanks for the push to try the latest git version of avrdude.  for me, that now fixed the problem.  please consider pulling in latest for arduino people so that this does not happen anymore.

sercona, 2022-04-29:
oh, and I always kill modemmanager and network-manager when I install linux.  (that pisses off one of my friends who is the modemmanager guy...) but for me, I never use it and on x64 hosts, its always off.

network-manager is another thing I hate having, but that's for another forum/thread.

sercona, 2022-04-29:
fwiw, it feels like a timing issue since it fails about 75% of the time.  it gets partially along with the series of progress chars, but then stops and times out.

in the past, there have been issues with other usb isochronous things when chipsets enter the linux world.  I had usb-audio problems when skylake came out (or in that time frame) and the previous chipset was perfect.

at any rate, the latest avrdude solves the issue.

for those that don't see the issue, see if the 'supplied' version from arduino (6.3-20190619) works well on your amd box.  for me, its a loser.

sercona, 2022-04-29:
sorry, one more bit of info that will be useful: when I went back to re-re-test this, the old arduino avrdude now works.

stranger and stranger.

unplug usb and replug (get usb0 again) and it works still.

something 'primed the pump' (sorry) by running the more modern version and now running either works.

so, debugging this is not going to be a simple thing.  it still feels like buffering or timing or transactions acting different and not according to what the protocol has been expecting.

I have not dug deeper, but almost reminds me of apps that set TTY characteristics and once those are set, other apps that use that com port can use that port with those settings still 'set', and maybe avrude (old version) works when some previous state things are set.  (total guess, probably wrong)

sercona, 2022-05-04:
I've had to reboot my main system (that was up for half a year or more) and I'm happy to say that continued use of the latest avrdude gives me 100% success from my ryzen box.

I find where avrdude (bin) is in the arduino ide run tree, I move it to .old and "ln -s" to where my built-from-git avrude lives (/usr/local/bin, likely).

its a pain that I'll have to remember to do this as I unpack new arduino versions from zip into my home folder, but this work-around will give me 100% flash ability to nano and likely also usb-flashed wemos d1 minis (that I use a ton of, and usb has to be done for the first one; OTA updates handle all after that but first one has to be usb, so usb really does need to work) ;)

what is it going to take to get arduino org to package a more recent version (if not latest) of avrdude?