arduino/Arduino #8277: Problem with Arduino 1.8.8 (Store Edition 1.8.19.0) compiling SPI

johannstieger, 2018-12-09:
Since the latest release I get this error message:

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\libraries\SPI\src\SPI.cpp: In static member function 'static void SPIClass::begin()':

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\libraries\SPI\src\SPI.cpp:64:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

exit status 1

cmaglie, 2018-12-09:
Is this happening after upgrading to 1.8.8?
Could you share the whole sketch?
Can you enable verbose compile output from preferences and copy the output here? (please use the button "Copy error messages" that appears on the orange bar after compiling).

johannstieger, 2018-12-09:
Hi Cristian,
yes, it it since the update of 1.8.8 since today.

[Node_Moteino_2018-12-08.txt]
(https://github.com/arduino/Arduino/files/2660717/Node_Moteino_2018-12-08.txt)

The ino is attaced as a txt-file. The error message you can see below.

Thanks, Johann

Arduino: 1.8.8 (Windows Store 1.8.19.0) (Windows 10), Board: "Moteino / MoteinoUSB"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Users\Johann\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Users\Johann\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Users\Johann\Documents\Arduino\libraries -fqbn=Moteino:avr:Moteino -ide-version=10808 -build-path C:\Users\johan\AppData\Local\Temp\arduino_build_861689 -warnings=none -build-cache C:\Users\johan\AppData\Local\Temp\arduino_cache_510221 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.0.1-arduino5.path=D:\Users\Johann\Documents\ArduinoData\packages\arduino\tools\avrdude\6.0.1-arduino5 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=D:\Users\Johann\Documents\ArduinoData\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose D:\Users\Johann\Documents\Arduino\Node_Moteino_2018-12-08\Node_Moteino_2018-12-08.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Users\Johann\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Users\Johann\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Users\Johann\Documents\Arduino\libraries -fqbn=Moteino:avr:Moteino -ide-version=10808 -build-path C:\Users\johan\AppData\Local\Temp\arduino_build_861689 -warnings=none -build-cache C:\Users\johan\AppData\Local\Temp\arduino_cache_510221 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.0.1-arduino5.path=D:\Users\Johann\Documents\ArduinoData\packages\arduino\tools\avrdude\6.0.1-arduino5 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=D:\Users\Johann\Documents\ArduinoData\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose D:\Users\Johann\Documents\Arduino\Node_Moteino_2018-12-08\Node_Moteino_2018-12-08.ino
Using board 'Moteino' from platform in folder: D:\Users\Johann\Documents\ArduinoData\packages\Moteino\hardware\avr\1.4.0
Using core 'arduino' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-ID:\\Users\\Johann\\Documents\\ArduinoData\\packages\\Moteino\\hardware\\avr\\1.4.0\\variants\\Moteino" "C:\\Users\\johan\ [... cut ...]
Using library SPI at version 1.0 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\libraries\SPI 
Using library RFM69_LowPowerLab at version 1.1.3 in folder: D:\Users\Johann\Documents\Arduino\libraries\RFM69_LowPowerLab 
Using library Low-Power at version 1.6 in folder: D:\Users\Johann\Documents\Arduino\libraries\Low-Power 
Using library ArduinoJson at version 5.13.3 in folder: D:\Users\Johann\Documents\Arduino\libraries\ArduinoJson 
exit status 1
Error compiling for board Moteino / MoteinoUSB.

johannstieger, 2018-12-09:
Hi Cristian,
you can ignore this issue - I think it was a windows-store-problem.
A restart of my machine solved the problem.

Thanks, Johann

closed 2018-12-09 by johannstieger

atilacska, 2018-12-12:
prečo mi vypisuje pri nahrávaní GRBL upload exit status 1?Dakujem za pomoc.