arduino/Arduino #8288: [Library Manager] please add Programmable-Air library

tinkrmind, 2018-12-11:
The library can be found here: [https://github.com/Programmable-Air/Code](https://github.com/Programmable-Air/Code)

The library is for use with Programmable-Air hardware which is essentially a shield for an Arduino Nano. More about the hardware at [https://github.com/Programmable-Air/PCB](https://github.com/Programmable-Air/PCB)

cmaglie, 2018-12-11:
You must put the content of `programmable_air` folder on the root folder of the git repository (together with LICENSE and README), this is a requirement for publishing the library on the library manager, remember also to redo the tag once done.

tinkrmind, 2018-12-11:
Thank you for the response. I have moved the files to the root directory and updated the tag.

cmaglie, 2018-12-18:
Still can't find `library.properties` on the `1.0.0` tag:

https://github.com/Programmable-Air/Code/tree/1.0.0

seems that you forgot to redo the tag on the master branch.

tinkrmind, 2018-12-24:
My bad! I thought I just had to edit the release. I have issued a new release here:
https://github.com/Programmable-Air/Code/tree/1.0.1
and edited library.properties accordingly.

cmaglie, 2019-01-11:
done!

closed 2019-01-11 by cmaglie

tinkrmind, 2019-01-13:
Yay!!