arduino/Arduino #8293: ArduinoISP will be reset regardless I choose "Upload with Programmer" in 1.8.8 and 1.8.7 , which stops it working when upload to breadboards or AVR chips.

hehongbo, 2018-12-12:
I used to download program into breadboard or single AVR chips with Arduino Uno and _ArduinoISP_ from example menu, with Uno's Pin-11,12,13 goes to target's SPI and pin-10 goes to target's RST# , It works fine with 1.8.5. 

But in 1.8.7, this function stops working. It seems Arduino IDE will reset Arduino board regardless it is actually an ISP, not the target chip I'm about to erase and download.

So to prevent this I need to **hook the reset header of Uno with a 10uF capacitor, which stops Uno's RST# from dropping low and being reset**. By doing so upload will succeed or I will receive the following error message otherwise: 

```
Arduino: 1.8.8 (Mac OS X), Board: "ATmega328, No, 328P / 328PA, 2.7v, Disabled (default), 16 MHz external"

Sketch uses 450 bytes (1%) of program storage space. Maximum is 32768 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x01
avrdude: stk500_initialize(): (a) protocol error, expect=0x14, resp=0x10
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): unknown response=0x12
the selected serial port avrdude: stk500_disable(): unknown response=0x12
 does not exist or your board is not connected

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

But I'm sure the 10uF capacitor on RST# is not necessary with "_Upload with Programmer_" function in 1.8.5.
I discovered this issue in 1.8.7 and I'm not sure whether it works under 1.8.6.
But in 1.8.8 it is not fixed.

thanks and merry X-mas.

facchinm, 2018-12-12:
Hi @hehongbo ,
the issue might be related with https://github.com/arduino/ArduinoCore-avr/issues/45 and https://github.com/arduino/Arduino/issues/8032 . 
It looks like everyone has a different problem so I'd ask you to try to change the protocol from `arduino` to `stk500v1` and report if it works. Thanks!

hehongbo, 2018-12-12:
Hi @facchinm 
Thank you! I diffed `programmers.txt` with another computer which still  runs 1.8.5 and noticed protocol changes from `stk500v1` to `arduino`. So I create a new programmer configuration with `stk500v1` and so glad to see it works again. I'm so appreciate.

closed 2018-12-12 by hehongbo