arduino/Arduino #8297: tone() used with delay()

cds333, 2018-12-13:
If the delay is less than the duration no tone will play. Using delay() after tone() only works if the delay is longer than the tone's duration, and then only for the time that is equal to (duration - delay)

For example, looping the following plays nothing:

tone(piezoPin, 2000, 500);
delay(500);

While looping the following plays 500ms tones followed by 50ms of silences

tone(piezoPin, 2000, 500);
delay(550);

Using official build 1.8.7

PaulStoffregen, 2018-12-13:
Please give us a complete program, so we can copy it into Arduino without having the fill in (guess) the rest of the code.

Also, let us know which board.  The tone() code is different on some boards, so that too is a pretty important detail for anyone to actually investigate this problem.

cds333, 2018-12-14:
void setup() {
}

void loop() {
     tone(piezoPin, 2000, 500);
     delay(500);
}

cds333, 2018-12-14:
Lanmu Arduino Uno R3 clone:

https://www.miniarduino.com/lanmu-uno-r3-board-atmega328p-atmega16u2-compatible-with-arduino-uno-r3/

Also tried it on another brand R3 clone and same thing.