arduino/Arduino #8305: When I use Arduino IDE 1.8.8, I always fail to upload programs

chen462591558, 2018-12-15:
I have a problem. My Windows system is 64-bit win10. When I use Arduino IDE 1.8.8, I always fail to upload programs, but I have no problem using Arduino IDE 1.6.5 upload programs. Is this the new IDE's support for 64-bit win10 not good? I hope the developers of Arduino IDE can fix this problem in time, or you can tell me how to solve this problem skillfully. Thank you. My email address is 462591558@qq.com.

per1234, 2018-12-15:
Please do this:
1. File > Preferences > Show verbose output during: > compilation (uncheck) > upload (check) > OK
1. Sketch > Upload
1. After the upload fails you'll see a button on the right side of the orange bar "Copy error messages". Click that button.
1. Paste the error messages in a reply here [using code fencing](https://guides.github.com/features/mastering-markdown/).

bjoc8com, 2018-12-15:
I use Arduino Uno as "Arduino as ISP" to upload.

per1234, 2018-12-16:
@bjoc8com you may be encountering https://github.com/arduino/Arduino/issues/8032. That can be solved in either of two different ways:

----
#### Fix A:
Connect a 10 uF capacitor between the GND and RESET pins on the board you're using as the "Arduino as ISP".

----
#### Fix B:
Roll back to Arduino AVR Boards 1.6.21:
1. Tools > Board > Boards Manager
1. Wait for the downloads to finish.
1. Click on "Arduino AVR Boards".
1. Select "1.6.21" from the drop down menu.
1. Click "Install".
1. Wait for installation to finish.
1. Click "Close".
1. Try uploading again.

bjoc8com, 2018-12-16:
Yes, both versions work. I use version A.
Thanks
Are these solutions temporary or final?

per1234, 2018-12-17:
@bjoc8com there is some discussion of reverting the change that made it more likely for you to need the capacitor (https://github.com/arduino/Arduino/issues/8032#issuecomment-446562762). You can subscribe to https://github.com/arduino/Arduino/issues/8032 to get notifications of any progress on that.

Even after that, there's no harm in continuing to use the capacitor.

As for fix B, I would recommend only using that as a temporary workaround for people who don't have a capacitor on hand since it's unfortunate to miss out on the improvements made in later releases of Arduino AVR Boards.

per1234, 2019-02-12:
Closing as duplicate of https://github.com/arduino/Arduino/issues/8032

closed 2019-02-12 by per1234