arduino/Arduino #8306: [Library Manager] Please add Minecraft Link library to the Library Manager

KlemenDEV, 2018-12-15:
I am asking if it would be possible to add our library Minecraft Link to the Arduino library manager? The repository of the library can be found on this URL: https://github.com/Pylo/MinecraftLinkArduino

Thank you in advance!

cmaglie, 2018-12-18:
I've added the library to the index, but there is a problem:

```
JOB 1919 - 2018/12/18 17:58:47 Scraping https://github.com/Pylo/MinecraftLinkArduino.git
JOB 1919 - 2018/12/18 17:58:49 Checking out tag: 1.0-beta1
JOB 1919 - 2018/12/18 17:58:49 Error syncing library: Error generating library from repo: Invalid character(s) found in minor number "0-beta1"
```

the version `1.0-beta1` is not semver compliant it should be `1.0.0-beta1`.
Please fix this by:
1) changing the version in library.properties
2) commit & push
3) remove and redo the 1.0-beta tag

Once done our indexer should publish the library automatically within an hour, if this doesn't happen please comment here so we can sort it out.

closed 2018-12-18 by cmaglie

KlemenDEV, 2018-12-20:
Thanks, it works now.