arduino/Arduino #8307: IDE:  two modest proposals

RootlessAgrarian, 2018-12-16:
In my version of IDE (OSX) I see two minor inconveniences;  imho it would look slicker and cooler, and be easier and more intuitive to use, if these were fixed.

1) in the scrolling text frame at bottom of the sketch window, we see orange text showing compile progress, error messages, etc.  During load of the sketch onto a Due, there are two phases:  pushing the sketch over to the Due then verifying it.  Both of these have (sort of!) a progress bar in the bottom "compile messages" frame.  This progress bar however is pretty much a fail, because it (a) does not animate on one line, curses style, and (b) does not auto scroll either!  So all you see is the first "bite" of percentage update, unless you scroll manually.  So I end up dragging that text up repeatedly to see the progress of the flash.

2) I have more than 10 tabs in my sketch and there seems to be no way to drag or scroll the tabs horizontally.  I expected that a scrollbar would appear at the top just under or above the tabs as soon as my tab count exceeded the window width.  It would be nice if there was one :-)  As it is, I either have to make my window inconveniently wide to see all my tabs, or choose tab names from the menu at upper right.  When I choose a tab name from the menu at upper right, the text of the code in that tab is displayed in the editing window BUT the tab name is not updated, so it's not clear which tab I am editing!  I end up editing with no highlighted tab displayed at the top.  That's a bit confusing.  I think the tab bar should scroll instantly sideways to display the name of the (menu) selected tab in the farthest right position.  But it would be even better to be able to scroll the tab bar right/left to find the tab I am looking for, much more intuitive than a dropdown menu.

Since these are the only two issues I can think of with the IDE, it's a pretty darned impressive piece of software and I have really been enjoying using it  (for the last 2+ years).

per1234, 2018-12-16:
>there seems to be no way to drag or scroll the tabs horizontally. I expected that a scrollbar would appear at the top just under or above the tabs as soon as my tab count exceeded the window width.

We already have a feature request for this: https://github.com/arduino/Arduino/issues/3543

> I think the tab bar should scroll instantly sideways to display the name of the (menu) selected tab in the farthest right position.

We already have a feature request for this: https://github.com/arduino/Arduino/issues/1193

RootlessAgrarian, 2018-12-20:
Thanks for the note.  I am downloading even now and will give it a try... Will it work if not installed in the canonical location (Applications) under OSX?  I hope?  I have not tried this before and would like to leave the installed working version unharmed.

facchinm, 2018-12-20:
@RootlessAgrarian I get your point; the "chrome-like" behaviour is something I wanted to introduce to avoid scrolling (your solution 2) as long as we could (until the number of tabs become huge). Our graphical "toolkit" is not very smart :slightly_smiling_face: 
Normally I test that patch with Marlin, that also have files with similar names, but I navigate using the keyboard (CTRL + TAB) or mouse wheel, and stop at the desired file based on the title (may not work as intended on OSX, now I get it).
There was some code to make the selected tab normal size, but navigating created some kind of "motion sickness" :slightly_smiling_face: 

Anyway, any new idea (no second/third/fourth level bar since it doesn't scale, neither tree-like folder navigator on the left since it completely change the paradigm) is very well accepted!

[image]