arduino/Arduino #8308: Crazy bug? in IDE 1.8.6-8

sergikboy, 2018-12-16:
HI. I have fail of compilation my project in Arduino IDE 1.8.6.-8.
This project successfully building in Arduino IDE 1.8.5 and previos.
Project compile for Arduino MEGA2560. My system is 64-bit win10 and win8.1 32-bit.
I cut from my project smallest part of code for testing. Please dont change it for first compilation.
The last compile error "exit status 1 expected '}' before 'void'" .
But if I rename function GostinStrobeLight() -> GostinStrobeLight1() in DrMega2.ino and LightManagement.ino compilation will be successfully.
or compilation will be succesfull if  delete empty 1-22 line .
Arduino IDE 1.8.5 have not this problem. 
[DrMega2.zip](https://github.com/arduino/Arduino/files/2683915/DrMega2.zip)

facchinm, 2018-12-17:
Hi @sergikboy,
the bug happens if more than 23 blank lines are inserted. I think it's a trivial bug on our side, I'll take a look tomorrow.

facchinm, 2018-12-18:
I checked for regression and this bug was indeed solved by https://github.com/arduino/arduino-builder/commit/491a06767ebe5d0b03b9ddf0485de565e00a976f . I'm going to close this issue and release a new arduino-builder binary soon

closed 2018-12-18 by facchinm