arduino/Arduino #8309: too many commented rows

marcolorenziml, 2018-12-16:
i get this error:
OpenLCD:79:1: error: expected identifier before 'void'

when i try to add one or more lines before 'void setup()' 
[OpenLCD.zip](https://github.com/arduino/Arduino/files/2683957/OpenLCD.zip)

. 
commented lines also raise same error. 

for example: eliminating row 22 (<excedingRow>) from commented block in attached 'openLCD.ino' file i can compilation works fine, without problems.
with this row, instead, the compiler raise an error

per1234, 2018-12-17:
The problem is the function prototypes are being incorrectly inserted inside the enum definition:
```c++
enum displayMode
{
  MODE_NORMAL, //No mode, just print
  MODE_COMMAND, //Used to indicate if a command byte has been received
  MODE_SETTING, //Used to indicate if a setting byte has been received
  MODE_CONTRAST, //First setting mode, then contrast change mode, then the value to change to
  MODE_TWI, //First setting mode, then I2C mode, then change I2C address
  MODE_RECORD_CUSTOM_CHAR, //First setting mode, then custom char mode, then record 8 bytes
  MODE_SET_RGB //First setting mode, then RGB mode, then get 3 bytes
#line 77 "C:\\Users\\per\\Desktop\\OpenLCD\\OpenLCD.ino"
void setup();
#line 108 "C:\\Users\\per\\Desktop\\OpenLCD\\OpenLCD.ino"
void loop();
#line 127 "C:\\Users\\per\\Desktop\\OpenLCD\\OpenLCD.ino"
void updateDisplay();
#line 434 "C:\\Users\\per\\Desktop\\OpenLCD\\OpenLCD.ino"
void clearFrameBuffer();
#line 443 "C:\\Users\\per\\Desktop\\OpenLCD\\OpenLCD.ino"
void displayFrameBuffer(void);
#line 8 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void changeIgnore();
#line 40 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void displayFirmwareVersion();
#line 57 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void changeContrast(byte contrast);
#line 79 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void changeTWIAddress(byte newAddress);
#line 100 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void changeSplashContent();
#line 119 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void changeBLBrightness(byte color, byte brightness);
#line 161 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void changeBacklightRGB(byte red, byte green, byte blue);
#line 178 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void changeUARTSpeed(byte setting);
#line 240 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void changeSplashEnable();
#line 268 "C:\\Users\\per\\Desktop\\OpenLCD\\Setting_Control.ino"
void changeLinesWidths(byte setting);
#line 31 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void serialEvent();
#line 49 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void twiReceive(int rxCount);
#line 66 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setupPower();
#line 87 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setupUART();
#line 115 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setupSPI();
#line 133 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setupTWI();
#line 150 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setupContrast();
#line 167 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setupLCD();
#line 202 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setupBacklight();
#line 221 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setupSplash();
#line 316 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setupCustomChars();
#line 333 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void checkEmergencyReset(void);
#line 388 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
long lookUpBaudRate(byte setting);
#line 410 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void petSafeDelay(int delayAmount);
#line 436 "C:\\Users\\per\\Desktop\\OpenLCD\\System_Functions.ino"
void setPwmFrequency(int pin, int divisor);
#line 66 "C:\\Users\\per\\Desktop\\OpenLCD\\OpenLCD.ino"
} currentMode = MODE_NORMAL;
```
The problem does not occur when using Arduino IDE 1.9.0-beta build 95 so this is fixed in [arduino-preprocessor](https://github.com/arduino/arduino-preprocessor). For that reason, I don't know whether it's worth pursuing a fix in the [arduino-builder](https://github.com/arduino/arduino-builder) tool, which is handling function prototype generation until arduino-preprocessor is ready for a production release of the Arduino IDE. I'll let one of the developers make that call.

The workaround is to manually add the function function prototypes in an appropriate location in the code. arduino-builder only generates prototypes for functions that don't already have one. So if you add the following code at line 76 of OpenLCD.ino, the sketch will compile:
```c++
void setup();
void loop();
void updateDisplay();
void clearFrameBuffer();
void displayFrameBuffer(void);
void changeIgnore();
void displayFirmwareVersion();
void changeContrast(byte contrast);
void changeTWIAddress(byte newAddress);
void changeSplashContent();
void changeBLBrightness(byte color, byte brightness);
void changeBacklightRGB(byte red, byte green, byte blue);
void changeUARTSpeed(byte setting);
void changeSplashEnable();
void changeLinesWidths(byte setting);
void serialEvent();
void twiReceive(int rxCount);
void setupPower();
void setupUART();
void setupSPI();
void setupTWI();
void setupContrast();
void setupLCD();
void setupBacklight();
void setupSplash();
void setupCustomChars();
void checkEmergencyReset(void);
long lookUpBaudRate(byte setting);
void petSafeDelay(int delayAmount);
void setPwmFrequency(int pin, int divisor);
```

facchinm, 2018-12-19:
I just reproduced it and it's the same bug as #8308, solve by the same commit. Closing here, will be definitely fixed by the next release of the builder.

closed 2018-12-19 by facchinm