arduino/Arduino #8313: Arduino IDE (sometimes( calls avrdude with wrong parameters

peteeckel, 2018-12-17:
I have a strange problem with the Arduino IDE on Mac OS X 10.14. 

I'm trying to program an Arduino Nano clone (Elegoo nano), of which I have several specimens which used to be working perfectly. 

Recently I upgraded my development machine to macOS 10.14 (Mojave) and the Arduino IDE to 1.8.8, and I found that I had troubles programming my Arduinos with a variety of different errors from avrdude. The problem turned out to be the IDE calling avrdude with the wrong parameters. 

Setup: 

Board: "Arduino Nano"
Processor: "AtMEGA 328P (Old Bootloader)"
Port: "/dev/wchusbserial1420"
Programmer: "AVRISP mkII"

Turning on verbose logging, I found that the IDE calls avrdude as follows:

`/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -cstk500v2 -Pusb -Uflash:w:/var/folders/yt/10tpqq355nsdfxkhw7y293zm0000gn/T/arduino_build_859586/Cyber_2018.ino.hex:i`

which is obviously wrong. When I run avrdude manually with the correct parameters as 

`/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -P/dev/cu.wchusbserial1420 -b57600 -D -Uflash:w:/var/folders/yt/10tpqq355nsdfxkhw7y293zm0000gn/T/arduino_build_859586/Cyber_2018.ino.hex:i`

uploading the code works perfectly. 

What makes the problem seem even more strange is that **sometimes** - without any changes - avrdude is called correctly, then again, just when I hit Command-U again without any changes, the wrong parameters creep in again.

per1234, 2018-12-17:
The "wrong parameters" output is the result of doing an "Upload Using Programmer" instead of a standard Upload. The keyboard shortcut for "Upload Using Programmer" is Command+Shift+U. Are you sure you're not pressing the shift button by accident?

peteeckel, 2018-12-17:
I'm pretty sure I did not (I'm very used to using keyboard shortcuts), but I can't say it with a 100% confidence ... of course the last few times I tried it worked perfectly. I'll keep a close eye on it and will close the ticket when it turns out that one of my fingers does things it shouldn't without me noticing it.

peteeckel, 2018-12-18:
Of course you were correct, and I was wrong ... obviously something crept into my muscle memory that had me consistently hit the wrong shortcut. After your reply made me aware of it, it never happened again - PEBCAK. That's a bit embarrassing. 

Thanks for your help!

closed 2018-12-18 by peteeckel