arduino/Arduino #8316: Please add Arduino-CloudStorage to the Library Manager

gilmaimon, 2018-12-18:
Hi!
Following instructions from: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
Please add my library "Arduino-CloudStorage" to the library manager. library available here: https://github.com/gilmaimon/Arduino-CloudStorage

Thank You!

per1234, 2018-12-18:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

gilmaimon, 2018-12-18:
> You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file. [...]

Thank You!
Ok so done, I think. Created release named 0.1 (matching the version in library.properties) for both repos. Does it look ok?
Cheers for the quick reply!