arduino/Arduino #8318: wrong processing of uint64

ricardomsmonteiro, 2018-12-19:


facchinm, 2018-12-19:
The disassembly shows that all the check is being stripped if **both** the shifted "immediate" are bigger than 8.
In fact, this works as expected and produces much more code.
```if( (foo.bar  & (1ULL << 1)) && (foo.bar  & (1ULL << 12)) )```
I'll investigate a little bit before reporting to gcc maintainers

facchinm, 2018-12-19:
Replacing `<< 9` with `<< 1` gives this disassembly output diff:
```
d82:	20 91 0c 01 	lds	r18, 0x010C	; 0x80010c <foo>
 d86:	22 70       	andi	r18, 0x02	; 2
 d88:	30 91 0d 01 	lds	r19, 0x010D	; 0x80010d <foo+0x1>
 d8c:	30 71       	andi	r19, 0x10	; 16
 d8e:	40 e0       	ldi	r20, 0x00	; 0
 d90:	50 e0       	ldi	r21, 0x00	; 0
 d92:	60 e0       	ldi	r22, 0x00	; 0
 d94:	70 e0       	ldi	r23, 0x00	; 0
 d96:	80 e0       	ldi	r24, 0x00	; 0
 d98:	90 e0       	ldi	r25, 0x00	; 0
 d9a:	22 30       	cpi	r18, 0x02	; 2
 d9c:	30 41       	sbci	r19, 0x10	; 16
 d9e:	41 05       	cpc	r20, r1
 da0:	51 05       	cpc	r21, r1
 da2:	61 05       	cpc	r22, r1
 da4:	71 05       	cpc	r23, r1
 da6:	81 05       	cpc	r24, r1
 da8:	91 05       	cpc	r25, r1
 daa:	b1 f4       	brne	.+44     	; 0xdd8 <main+0x198>
```
(only in the correct execution).
It looks like the compiler is stripping a (const?) check it think is always false. I'll go ahead and produce a very minimal test case before reporting in mainline.