arduino/Arduino #8322: [Library Manager] Add Dimmable Light (or generic AC loads) library

fabiuz7, 2018-12-19:
Hi all,
Could you add this library to Arduino Library Manager?https://github.com/fabiuz7/dimmable_light_esp8266_lib
In the next days I'm going to add support also for original Arduino UNO.

Thanks

per1234, 2018-12-20:
The `name` value in your library.properties contains a prohibited character: `/`. Please change the `name` value to contain only basic letters (A-Z or a-z) and numbers (0-9), spaces ( ), underscores (_), dots (.) and dashes (-). After that, you will need to create a new tag/release or else redo the existing tag.

Once that's done, please comment here so we'll know we can proceed.

Reference:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

fabiuz7, 2018-12-21:
Sorry, I forgot to re-check the specifications... Now it should be ok!

cmaglie, 2018-12-24:
I've added the library to the index but there is a problem:
```
JOB 1924 - 2018/12/24 12:05:00 Scraping https://github.com/fabiuz7/dimmable_light_esp8266_lib.git
JOB 1924 - 2018/12/24 12:05:03 Checking out tag: 1.0.0
JOB 1924 - 2018/12/24 12:05:03 Error syncing library: Error generating library from repo: 'sentence' and 'url' fields must be defined
JOB 1924 - 2018/12/24 12:05:03 Checking out tag: 1.1.0
JOB 1924 - 2018/12/24 12:05:03 Error syncing library: Error generating library from repo: 'sentence' and 'url' fields must be defined
```

this happens because the `url` field is empty, to fix it:
1) edit `library.properties` and add a value to the `url` property
2) commit and push
3) delete and redo the latest release tag

once done the library will be automatically published within an hour, no need to comment further here.

closed 2018-12-24 by cmaglie