arduino/Arduino #8323: [Proposal] Declare library depencies inside sketches

philippejadin, 2018-12-20:
This proposal is short and adress a shortcoming for every user : which library do I need to install to make this sketch run?

- allow to add some markup to sketches to declare the required library and version needed 
- parse this new info on compile, and ask user if he wants to install the needed libraries
- profit!

**Markup proposal, example with the rfid lib :** 
```
// require_arduino_lib "MFRC522"
#include <MFRC522.h>
```

Note that we simply use a comment so nothing changes from the compiler point of view.

**Markup proposal specifyng the lib version to use :** 
```
// require_arduino_lib "MFRC522" "1.4.2"
#include <MFRC522.h>
```

philippejadin, 2019-01-01:
Sorry to bump this thread, but really no one is interested by this? It's such a common problem to install the correct library for newbies... I see a lot of potential in my proposal but maybe I missed something :-)

facchinm, 2019-01-02:
Hi @philippejadin ,
the proposal is interesting; however, there's a lot of infrastructure to be modified before approaching this (first of all, allowing two different versions of the same library to coexist).
In the meantime, there's a less known feature to directly open library manager that could be useful; a link in the form `http://librarymanager/all#keywordToSearch` (inside a comment) will open the library manager with the right filters and ease the installation at least.

scls19fr, 2019-01-03:
Looks related to https://github.com/arduino/Arduino/issues/5795

ericvb, 2020-04-26:
@philippejadin Have you been able to make progress on this?
This would boost the usability of sketches if you could include in the script the dependencies to libraries and the used version. 
Everybody would then be able to compile your script without a problem of finding the right libs.

This of course requires that all library versions remain available indefinitely...

philippejadin, 2020-04-27:
Unfortunately all I have done is make a proposal. This must be decided by the core team and implemented by very competent people (which does not include me in both cases :-) )

philippejadin, 2020-04-27:
This being said, I still think that my proposal holds well and is simple enough for users / code writers.
BUT : it must be supported by the IDE, that is the hard part, well worth it imho.

ericvb, 2020-04-27:
@facchinm Do you know if there is something on the roadmap for this problem?

Your suggestion to use a url link to the librarymanager could already help, but it is very limited.
- the url link search works not for library names with spaces (also not in the IDE itself)
- if it should be able to include the required library version
1. no version set ==> take the latest version and install it
2. a version set ==> take that version and install it
3. a minimum version set ==> install at least this minimum version if the lib is already installed on the system or take the latest version if not installed

This latest point could be implemented without requiring a change of the Arduino infrastructure.