arduino/Arduino #8339: error: espcomm_upload_mem failed Let's Start Coding Spark Fun ESP8266 Thing

BoeingLikesToDab, 2018-12-28:
When I try to upload a project to my Let's Start Coding SparkFun 

ESP8266 Thing board, it gives me this: 
[image]
It was just a test, so the code is fairly simple, here it is without the stupid screenshot hard to see-ness:
`
//random arduino test
void setup(){
pinMode(12,OUTPUT);
}

void loop(){
digitalWrite(12,HIGH);
delay(2000);

digitalWrite(12,LOW);
delay(2000);
}
`

BoeingLikesToDab, 2018-12-28:
If it matters, I am working on a 2017 Surface Pro with an i5-7300U with Intel HD 620 and 8GB of RAM. I have a 256GB Toshiba SSD with Windows 10 Pro

per1234, 2018-12-28:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2018-12-28 by per1234

BoeingLikesToDab, 2018-12-28:
Ok