arduino/Arduino #8341: [Proposal] Provide for hook into Timer0 overflow ISR

gbhug5a, 2018-12-29:
It would be useful if there was a way to add additional operations or functions to the existing timer0 ISR.  Using  some other timer interrupt might conflict with normal functions, such as analog writes.  But simply adding code to the existing ISR conflicts with nothing.  The 1 ms rate, or a multiple of it, would be perfect for polling functions, such as servicing a rotary encoder or momentary switches, or display refreshes, in a way that wouldn't be compromised by delay() or other loop() operations.  Perhaps the existing ISR could call or jump to a specifically named function, if it exists.  Or if there was a way to replace the interrupt vector with mine, I could write a new ISR which includes the standard millis update code in addition to my other functions.

matthijskooijman, 2018-12-29:
Interesting idea. A weak function could work here, or just a function pointer called through a global variable perhaps.

> The 1 ms rate, or a multiple of it,

That's not entirely accurate: IIRC timer0 on AVR runs at 1024μs, not 1000μs. That would probably reduce the usefulness of such an interrupt hook, though still work for a lot of cases.

An API for this would probably not be portable to other architectures (or at least the interval is likely to be different on other architectures).

matthijskooijman, 2018-12-29:
> My understanding of designating weak functions is close to zero, but my impression is that that would need to be done somewhere in the core.

Correct. I meant this as a suggestion for a possible implementation in the core, not for something you could do already :-)

> My understanding of designating weak functions is close to zero, but my impression is that that would need to be done somewhere in the core.

Unfortunately, the MCU cannot manipulate memory directly, so globals have to be loaded into registers, then changed, then written back to memory. However, the compiler already completely handles the pushing and popping of registers. You do make a good point, though, since *if* you call a function inside an ISR, a *lot* more registers might be clobbered and must be preserved, which likely means the ISR gets a lot slower (even when not actually calling a function). You could circumvent this by marking the called function as an ISR as well, making it preserve all registers it uses itself, but that might be a lot more tricky for the user (easier to forget and break things...)

gbhug5a, 2018-12-29:
Do you mean that even though a global variable is assigned a RAM address, the processor isn't allowed to modify its value directly?  Well that's confusing.  I'm trying to remember if any of the examples I've seen of setting up timer1 or timer2 interrupts and ISRs bothered to push/pop anything.  I don't think so.  But it sounds like dealing with variables and registers could be an issue.

Well let me ask this.  I've written code for a reflow controller that includes 7-segment multiplexing by segment, a rotary encoder with its push button, and a buzzer.  I wrote it orignally in assembler for MSP430, and the whole thing is just a big polling routine in the ISR, servicing one thing after another.  The processor is literally turned off at all other times.  But I've since ported that to Energia, which for this issue is probably equivalent to Arduino, and for that I moved everything into loop().  It just waits until millis have incremented by two, then executes the code.  It needs to be 2 or 3 millis to keep the displays from flickering, and to properly service the encoder.  I'm currently awaiting delivery of Arduinos, and will then transfer from Energia to Arduino (hopefully just changing the pin assignments).

Suppose I set up a timer2 ISR and put the whole thing in there, and made loop() empty.  Would that tell us anything about how well the registers and variables might work, or whether it's slower there than in loop()?  The Energia compile says global variables take up 142 bytes of ram, most of which are byte variables, so this would be a substantial ISR.   I guess it would be better to also have something running in loop() to see if there's any conflict between that and the ISR.  Well, let me know if you think that would be a useful exercise or not.

matthijskooijman, 2019-01-03:
> Do you mean that even though a global variable is assigned a RAM address, the processor isn't allowed to modify its value directly? Well that's confusing.

Correct. It's not so much "not allowed to", but "cannot", since it simply has no instructions for this. I think this is a common way to build processors (and it makes sense, but I won't go into detail here).

As for the rest of your post, that is crossing from the "improve Arduino" subject into a "help with my project" subject, for which this is not the right place. Let me suffice by saying that your reasoning sounds ok to me and the proposed timer0 hook would indeed make something like this a bit easier and more accurate than loop-based or even timer2-based solutions. However, if you do not have anything else to do in the loop, running polling checks in the loop is probably also just as fine (unless you really need to be super acurate or jitter-free, which I suspect you don't).