arduino/Arduino #8344: 1.8.8 Arduino IDE Windows Installer drops avrdude.conf in incorrect directory

automaticjack1, 2018-12-30:


per1234, 2018-12-30:
Please do this:
1. File > Preferences > Show verbose output during: > compilation (uncheck) > upload (check) > OK
1. Sketch > Upload
1. After the upload fails you'll see a button on the right side of the orange bar "Copy error messages". Click that button.
1. Paste the error messages in a reply here [using code fencing](https://guides.github.com/features/mastering-markdown/).

automaticjack1, 2018-12-30:
The avrdude.conf file is installed here:

C:\Users\USER\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\bootloaders\gemma\avrdude.conf

The global configuration expects the avrdude configuration file to be here:

C:\Users\USER\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc

This directory does not exist with a clean installation from the Windows installer.

The workaround is to create the directory and copy the existing avrdude.conf to it. This immediately results in the ability to upload sketches to the attached Arduino devices.

It's not clear what the long-term fix is for this.

The system is Windows 10. It is unlikely that other OSes will have the same issue. The zip file version has not been tested or inspected for this issue.

automaticjack1, 2018-12-30:
Arduino: 1.8.8 (Windows 10), Board: "Arduino Nano, ATmega328P"

Sketch uses 930 bytes (3%) of program storage space. Maximum is 30720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf -v -patmega328p -carduino -PCOM8 -b115200 -D -Uflash:w:C:\Users\jad\AppData\Local\Temp\arduino_build_484494/Blink.ino.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf"
avrdude: can't open config file "C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "C:\Users\jad\AppDatavrdude: error reading system wide configuration file "C:\Users\jad\AppDat
a\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf"

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

automaticjack1, 2018-12-30:
`Arduino: 1.8.8 (Windows 10), Board: "Arduino Nano, ATmega328P"

Sketch uses 930 bytes (3%) of program storage space. Maximum is 30720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf -v -patmega328p -carduino -PCOM8 -b115200 -D -Uflash:w:C:\Users\jad\AppData\Local\Temp\arduino_build_484494/Blink.ino.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf"
avrdude: can't open config file "C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "C:\Users\jad\AppDatavrdude: error reading system wide configuration file "C:\Users\jad\AppDat
a\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf"

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

automaticjack1, 2018-12-30:
[avrdude.txt](https://github.com/arduino/Arduino/files/2716877/avrdude.txt)

automaticjack1, 2018-12-30:
> Arduino: 1.8.8 (Windows 10), Board: "Arduino Nano, ATmega328P"

Sketch uses 930 bytes (3%) of program storage space. Maximum is 30720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf -v -patmega328p -carduino -PCOM8 -b115200 -D -Uflash:w:C:\Users\jad\AppData\Local\Temp\arduino_build_484494/Blink.ino.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf"
avrdude: can't open config file "C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "C:\Users\jad\AppDatavrdude: error reading system wide configuration file "C:\Users\jad\AppDat
a\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf"

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

automaticjack1, 2018-12-30:
Once the avrdude.conf is copied, the next attempt at uploading succeeds:

`Sketch uses 930 bytes (3%) of program storage space. Maximum is 30720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf -v -patmega328p -carduino -PCOM8 -b115200 -D -Uflash:w:C:\Users\jad\AppData\Local\Temp\arduino_build_484494/Blink.ino.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\jad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf"

         Using Port                    : COM8
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\jad\AppData\Local\Temp\arduino_build_484494/Blink.ino.hex"
avrdude: writing flash (930 bytes):

Writing | ################################################## | 100% 0.38s

avrdude: 930 bytes of flash written
avrdude: verifying flash memory against C:\Users\jad\AppData\Local\Temp\arduino_build_484494/Blink.ino.hex:
avrdude: load data flash data from input file C:\Users\jad\AppData\Local\Temp\arduino_build_484494/Blink.ino.hex:
avrdude: input file C:\Users\jad\AppData\Local\Temp\arduino_build_484494/Blink.ino.hex contains 930 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.34s

avrdude: verifying ...
avrdude: 930 bytes of flash verified

avrdude done.  Thank you.

`

per1234, 2018-12-30:
>The global configuration expects the avrdude configuration file to be here:
>
>C:\Users\USER\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc
>
>This directory does not exist with a clean installation from the Windows installer.

How much of that directory structure did exist? The Arduino IDE comes with the full installation of Arduino AVR Boards bundled in the IDE installation folder, including AVRDUDE. In a fresh installation, the location of avrdude.conf that will be used is {Arduino IDE installation folder}/hardware/tools/avr/etc/avrdude.conf. AVRDUDE is only installed to C:\Users\USER\AppData\Local\Arduino15\packages\arduino\tools\avrdude when you do an update of Arduino AVR Boards via **Tools > Board > Boards Manager** (or some other 3rd party hardware package that has a dependency on the AVRDUDE tool). It seems like you might have had an incomplete installation of an AVR package installed on your system before you did this installation of the Arduino IDE. The workaround for this class of problem we recommend on the Arduino forum for this class of problem is to delete the C:\Users\USER\AppData\Local\Arduino15 folder. In your case, that's probably not necessary since you seem to have found your own, less invasive, solution.

I'll leave it to the Arduino developers to determine whether they want to investigate if the IDE can be improved to avoid this sort of problem in the future, or whether it should be treated as a rare glitch that should just be worked around when it does occur..

automaticjack1, 2018-12-30:
AFAICT, the installation using the binary arduino-1.8.8-windows.exe created everything up to but not including the "exe" directory, thus:

    C:\Users\USER\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14

I had completely removed the installation directory (C:\Program Files (x86)\Arduino), but not touched the AppData directory.

Also, the MD5 hash of the exe installer file is this, in case anyone sneakily deploys a fix:

    c56c9511d9f0a519a9db5d795a99756a *arduino-1.8.8-windows.exe

facchinm, 2019-01-02:
All files/folders under `C:\Users\USER\AppData\Local\Arduino15\packages` are only created by Board Manager, not by the exe installer. You probably have a previous installation (via board manager) that didn't end up well and the IDE is trying to use that instead than the bundled one.
My suggestion is: start from scratch by removing the whole `C:\Users\USER\AppData\Local\Arduino15` folder, retry the upload (it should use the bundled tools) and let us know if it works.

per1234, 2019-03-26:
@automaticjack1 did you ever get a chance to try what facchinm recommended in the last reply? If so, did it solve your issue?

automaticjack1, 2019-03-26:
No, I had not. For some reason I never received notification about facchinm's request.

I did work around the issue by copying the necessary configuration file into the place the tool expected it, and have had no issues since.

If there are no other parallel reports of this type, then this issue is probably a one-off or unique occurrence. I haven't had issues with 1.8,8 on either machine I use it on since. The installer also didn't generate any warnings about not being able to create files at the time it was run.

Is there an installer log I might be able to dig up for you?

ivandemec, 2021-11-29:
For me an easy fix was to simply copy the file ```avrdude.conf```  from the installation directory ```C:\Users\IvanDemec\Downloads\arduino-1.8.16-windows.zip\arduino-1.8.16\hardware\tools\avr\etc``` to the directory ```C:\Users\IvanDemec\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17\etc```
After that it works fine.
This happened to me while trying, through board manager, to update AVR boards library.

PrasennaPS, 2023-11-02:
Sketch uses 3796 bytes (11%) of program storage space. Maximum is 32256 bytes.
Global variables use 190 bytes (9%) of dynamic memory, leaving 1858 bytes for local variables. Maximum is 2048 bytes.
"C:\Users\Prasenna\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\Prasenna\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -V -patmega328p -carduino "-PCOM16" -b115200 -D "-Uflash:w:C:\Users\Prasenna\AppData\Local\Temp\arduino\sketches\9BA35EB3DE4BB811449C5947C2B55E94/Car_Parking_System (2).ino.hex:i"

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Prasenna\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"
avrdude: can't open config file "C:\Users\Prasenna\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "C:\Users\Prasenna\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"
Failed uploading: uploading error: exit status 1

can anyone say how to solve this??