arduino/Arduino #8360: How many .ino files are allowed in the sketch folder?

mitag, 2019-01-04:
Under Windows 10 only the Sketch is allowed to use the .ino extension, other files with this extension are ignored by the compiler. Tested with version 1.8.7. I can compile the same project under Debian Linux with version 1.8.5 without error messages. Since this is not my project, I have also contacted the developers, but no answer yet. Of course I don't know if they work with a Mac or Linux. I also don't know if the compiler will work under Windows or Linux. [OpenLCD](https://github.com/sparkfun/OpenLCD/tree/master/firmware/OpenLCD)

mitag, 2019-01-04:
I'm sorry, I mutilated the last sentence, which should be:  I also don't know if the compiler will work as desired under Windows or Linux.

per1234, 2019-01-06:
It's caused by https://github.com/arduino/Arduino/issues/8308. It's not a problem of multiple .ino files. It's a problem of the automatically generated function prototypes being inserted inside the dataBuffer definition:
```c++
struct dataBuffer
{
  byte data[BUFFER_SIZE];  // THE data buffer
#line 70 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\OpenLCD.ino"
void setup();
#line 101 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\OpenLCD.ino"
void loop();
#line 120 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\OpenLCD.ino"
void updateDisplay();
#line 427 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\OpenLCD.ino"
void clearFrameBuffer();
#line 436 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\OpenLCD.ino"
void displayFrameBuffer(void);
#line 8 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void changeIgnore();
#line 37 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void displayFirmwareVersion();
#line 54 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void changeContrast(byte contrast);
#line 74 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void changeTWIAddress(byte newAddress);
#line 94 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void changeSplashContent();
#line 113 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void changeBLBrightness(byte color, byte brightness);
#line 155 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void changeBacklightRGB(byte red, byte green, byte blue);
#line 172 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void changeUARTSpeed(byte setting);
#line 234 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void changeSplashEnable();
#line 262 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\Setting_Control.ino"
void changeLinesWidths(byte setting);
#line 31 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void serialEvent();
#line 49 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void twiReceive(int rxCount);
#line 66 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setupPower();
#line 87 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setupUART();
#line 115 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setupSPI();
#line 133 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setupTWI();
#line 150 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setupContrast();
#line 167 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setupLCD();
#line 202 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setupBacklight();
#line 218 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setupSplash();
#line 311 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setupCustomChars();
#line 328 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void checkEmergencyReset(void);
#line 383 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
long lookUpBaudRate(byte setting);
#line 405 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void petSafeDelay(int delayAmount);
#line 431 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\System_Functions.ino"
void setPwmFrequency(int pin, int divisor);
#line 66 "C:\\Users\\per\\Desktop\\OpenLCD-master\\firmware\\OpenLCD\\OpenLCD.ino"
  volatile byte head;  // store new data at this index
  volatile byte tail;  // read oldest data from this index
} buffer;  // our data buffer is creatively named - buffer
```
This bug has already been fixed in the [hourly build of the Arduino IDE](https://www.arduino.cc/en/Main/Software#hourly).

In the future, please test with the hourly build before reporting bugs here so you can be sure they haven't already been fixed.

Related: https://github.com/sparkfun/OpenLCD/issues/15

closed 2019-01-06 by per1234

mitag, 2019-01-06:
I'm really sorry about that, but I spent more than an hour searching the issues to see if the bug is already known, I also tried various filters, but I didn't look for the description "Crazy bug". Thank you for your effort.

per1234, 2019-01-06:
Thanks for searching the issues for previous reports. I can sympathize with how difficult it can be to find anything in as many issues as this repository has. I don't blame you for not finding that one since it likely wouldn't have had any of the specific keywords you were searching for.