arduino/Arduino #8367: [Library Manager] Please add libraries

hideakitai, 2019-01-06:
Hello, I created some libraries. 
Could you add following to a list of libraries ?

StopWatch
https://github.com/hideakitai/StopWatch
XBeeATCmds
https://github.com/hideakitai/XBeeATCmds
Debunker
https://github.com/hideakitai/Debouncer
CRC
https://github.com/hideakitai/CRC
Filters
https://github.com/hideakitai/Filters
Convert
https://github.com/hideakitai/Convert
Debug
https://github.com/hideakitai/Debug
I2CExtension
https://github.com/hideakitai/I2CExtension
RingBuffer
https://github.com/hideakitai/RingBuffer
VectorXf
https://github.com/hideakitai/VectorXf

Let me know if this issue should be divided to one issue per one repository…

Thanks.

per1234, 2019-01-07:
There are a few issues that will prevent your libraries from being added to the Library Manger index:

#### Duplicate name
Each library in the Library Manager index must have a unique `name` value in library.properties. A couple of your libraries do not:
- RingBuffer: There is already a library of this name in the Library Manager index.
- Filters: The `name` value of this library is "CRC", which is the same as your other library of that name. I suspect this was a copy/paste error and you meant to name it "Filters.

#### Missing tags
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

---
After resolving the above issues, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

hideakitai, 2019-01-07:
@per1234 Thank you for checking! I'm sorry I forgot to release libraries and I fixed everything. Please check it again. Thanks!

hideakitai, 2019-01-07:
@per1234 Could you add one more library, please?

SPIExtension
https://github.com/hideakitai/SPIExtension

Thanks in advance!

per1234, 2019-01-07:
OK, that library will be added with the rest when cmaglie next goes through the list of outstanding Library Manager inclusion requests. The issue will be closed at that time and you'll get a notification from GitHub.

hideakitai, 2019-01-07:
I got it, thank you for your cooperation.

cmaglie, 2019-01-11:
done!

closed 2019-01-11 by cmaglie