arduino/Arduino #8376: Linux Sketch serial port permission erroneous error

rvsjimbo, 2019-01-10:
Arduino IDE for Linux V1.8.8 running on Linux Mint 19.1.

Plugged the Arduino in and it comes up on /dev/ttyACM0 in group dialout. Added myself to group dialout, and cat /etc/hosts >/dev/ttyACM0 runs without error, therefore I have write access to the port.

Uploading using the Arduino IDE generates the following error (which is obviously incorrect):

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/paul/Downloads/arduino-1.8.8/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/paul/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "/dev/ttyACM0": Permission denied

avrdude done.  Thank you.

Now, whilst there may be an error somewhere, permissions are definitely not the problem.

Best Regards,

Paul.

facchinm, 2019-01-10:
Hi @rvsjimbo ,
did you logout after adding to the dialout group? Probably the `echo` was working because you where in the same shell, otherwise for the setting to be applied system-wide you need to login again (or simply reboot)

facchinm, 2019-01-10:
Mmmmh, that's funny. The other thing that might be interfering is `modemmanager`, which you can safely uninstall if you don't use a 56k modem, since it tries to access any virtual serial port to check if it's a modem.

matthijskooijman, 2019-01-10:
I have indeed also seen permission denied errors, though they usually only appear within the first few seconds of plugging in a board (and I suspect they are indeed related to modemmanager). Note btw that modemmanager is mostly used for "dialing" with USB 3G modems (and possibly also for tethering to phones over bluetooth or USB). I'm not even sure it can do old skool 56k dialup at all :-)

Uninstalling modemmanager is certainly the easiest way to fix issues it causes. Adding rules to udev to make modemmanager ignore Arduino devices can work if you want to keep modemmanager around.

Still, none of this is necessarily related to modemmanager. But uninstalling it might be a good way to see if it is responsible (I'm not sure stopping it would be enough, could be it runs on udev hotplug...).

rvsjimbo, 2019-01-11:
Thats it! The rebels are there!

Yup, modemmanager was the problem. Removing that was the problem - thanks very much both of you for helping :-)

Paul.

closed 2019-01-11 by per1234