arduino/Arduino #8377: [Library Manager]Please add the Dabble Library

mimansamaheshwari, 2019-01-10:
Please add the Dabble Library to Library Manager - https://github.com/STEMpedia/Dabble
Tag - https://github.com/STEMpedia/Dabble/tags

per1234, 2019-01-10:
@mimansamaheshwari did you close this issue because you no longer wish to have your library added to the Arduino Library Manager index?

mimansamaheshwari, 2019-01-10:
Yes there are some edits in library that are suppose to be done before
publishing
it on Library Manager. Should I create the new issue or reopen the same?

per1234, 2019-01-10:
OK, that's fine. I just wanted to make sure it hadn't been done in error. Sometimes people think that just because I added the label, it means the library was already in the index, but that is done by another member of the Arduino team.

When you're ready, you can just reopen this issue.

mimansamaheshwari, 2019-01-10:
Ok thanks a lot for cooperation.

mimansamaheshwari, 2019-01-11:
Please publish this library to Arduino Library manager - https://github.com/STEMpedia/Dabble
Tag - https://github.com/STEMpedia/Dabble/tags

cmaglie, 2019-01-11:
done!

closed 2019-01-11 by cmaglie