arduino/Arduino #8378: [Library Manager] Please update TridentTD_LineNotify library new version 3.0

TridentTD, 2019-01-10:
Hi, Please add this library to the Library Manager:
https://github.com/TridentTD/TridentTD_LineNotify

Tag: https://github.com/TridentTD/TridentTD_LineNotify/releases/tag/3.0

Thanks

cmaglie, 2019-01-11:
Once the library is added there is no need to ask again for future releases: our indexer will publish automatically, every hour, all new releases made to the library.

BTW the release 3.0 is not yet published because you forgot to update library.properties:
https://github.com/TridentTD/TridentTD_LineNotify/blob/3.0/library.properties

in fact the version inside the file is still 2.2. To fix this:

1. update library.properties
2. commit & push
3. remove and redo the tag 3.0

TridentTD, 2019-01-13:
Already update version in library.properties to 3.0

https://github.com/TridentTD/TridentTD_LineNotify/blob/master/library.properties

Thank you.

per1234, 2019-01-13:
You still forgot to update the `version` value in library.properties before making the tag. The `version` value at your 3.0 tag is still `2.2`:
https://github.com/TridentTD/TridentTD_LineNotify/blob/3.0/library.properties

TridentTD, 2019-01-13:
OK, already update this link too.

https://github.com/TridentTD/TridentTD_LineNotify/blob/3.0/library.properties

TridentTD, 2019-01-25:
Hi @cmaglie

I have done by your advise
1. remove the tag 3.0
2. remove branch 3.0
3. make tag 3.0 again on master

per1234, 2019-01-25:
You need to update the `version` value in library.properties **before** making the tag. Currently, the `version` value in your 3.0 tag is still 2.2:
https://github.com/TridentTD/TridentTD_LineNotify/blob/3.0/library.properties#L2

TridentTD, 2019-01-28:
Sorry. 

OK. Updated to 3.0
https://github.com/TridentTD/TridentTD_LineNotify/blob/3.0/library.properties#L2

cmaglie, 2019-01-29:
Ok, this has been sorted out.

closed 2019-01-29 by cmaglie