arduino/Arduino #8380: [Library Manager] Please add new library

GordonRudman, 2019-01-11:
Please add https://gordonrudman-opensource.visualstudio.com/ExampleArduinoLibrary/_git/ExampleArduinoLibrary to the library list.

Thanks

per1234, 2019-01-11:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

GordonRudman, 2019-01-11:
Thanks. 
Sorry I missed that.
I've added the tag (0.0.1) now.

cmaglie, 2019-01-11:
@GordonRudman we only support repositories on github, bitbucket or gitlab, you must move the library to one of those services...

GordonRudman, 2019-01-12:
OK, no problem. I've changed the repo to:
https://github.com/GordonRudman/ExampleArduinoLibrary.git 

I would suggest that you start looking into supporting Microsoft Azure DevOps git repos too, as they are major/stable git repos (and offer developers many more tools). Microsoft actually even owns GitHub at this point.

PS: Thanks for the quick responses.

GordonRudman, 2019-01-15:
Hi @per1234 @cmaglie , 

I've noticed that this library isn't available through the Arduino Library Manager yet.
So, I was just wondering if there is anything else I need to do to?

Thanks,
Gordon

per1234, 2019-01-15:
You've now done everything required on your end. Now you only must wait.

Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

GordonRudman, 2019-01-15:
Great. Thanks.
Sorry for all the questions, this is just the first time I'm trying to create a public library.
I appreciate all the help.

cmaglie, 2019-01-21:
Done

closed 2019-01-21 by cmaglie