arduino/Arduino #8385: Code in documentation causes confusion

nickpotafiy, 2019-01-13:
Arduino often calls static methods, such as Wifi.begin(). Static methods are prevalent. When reading the documentation on [Arduino's WiFiSendReceiveUDPString page](https://www.arduino.cc/en/Tutorial/WiFiSendReceiveUDPString), if you don't pay attention to the declaration "WiFiUDP Udp" you will assume that the "Udp" in the code samples is a static class.

Not a problem though, if you don't declare "WiFiUDP Udp", you will just get an error that variable is missing, right? No. You get an error that Udp was renamed to EthernetUdp. And now you're off on a wild goose chase trying to figure out just what on earth EthernetUdp is and where do you need to get that library.

My suggestion is simple. Either change the documentation and set "WiFiUDP udp", thus changing each call to "udp.method", or remove the error that says Udp has been changed to EthernetUdp and provide the simple error that the variable is not declared. I think this will save a lot of headache for a lot of those starting off.

matthijskooijman, 2019-01-13:
> No. You get an error that Udp was renamed to EthernetUdp

Huh? Can you show the full error that you get? I'm wondering how that error is generated, I'm not aware of any compiler errors about "renaming" variables?

nickpotafiy, 2019-01-13:
Try it yourself. Here's all you need:

```
void setup() {
  Udp.endPacket();
}

void loop() {

}
```

And you get the following confusing error:

```
sketch_jan13b:3:3: error: The Udp class has been renamed EthernetUdp.
As of Arduino 1.0, the Udp class in the Ethernet library has been renamed to EthernetUdp.

   Udp.endPacket();

   ^

exit status 1
The Udp class has been renamed EthernetUdp.
```

facchinm, 2019-01-14:
This is a problem with the "fancy errors" that are handled by the IDE trying to be friendly with the user (see https://github.com/arduino/Arduino/blob/b5bfe08bd2c6f45d6762fad4cc9337063406d92a/arduino-core/src/cc/arduino/Compiler.java#L554-L557)
Some of them don't even apply anymore or are related with 0.x -> 1 transition (which happened eons ago) so getting rid of them shouldn't harm too much.
@cmaglie what should we do?

cmaglie, 2019-01-16:
I guess it's time to remove them.