arduino/Arduino #8391: Need a way to automate building and uploading a SPIFFS image from the CLI

scottchiefbaker, 2019-01-14:
I'm working on polishing a `Makefile` for automating compilation and uploading to an ESP8266 board. Using the Arduino IDE I'm able to build and upload a SPIFFS image per these [instructions](http://esp8266.github.io/Arduino/versions/2.0.0/doc/filesystem.html). I am unable to find a way to automate this from the [command line](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc). I understand that this is an external tool, but the Arduino IDE must be calling some CLI function to perform this task. 

What commands does the IDE use to compile and upload a SPIFFS image?

matthijskooijman, 2019-01-14:
If you're trying to figure out how to call the right commands without involving the IDE, this is not the right place to ask. I suggest you run with verbose compilation and uploading enabled, look at the platform.txt file for the esp8266 or ask on the github repo.

If you're asking to extend the IDE commandline options to allow calling the IDE to do the upload, then this *is* the right place. However, it seems that this uploading happens by installing a tool/plugin into the java IDE to call the right commands for the upload. I do not think we currently have any way to let such external tools add commandline options or actions, so this would require signfifcant refactoring of the IDE codebase, which I think would not be worthwile for just this particular usecase.

scottchiefbaker, 2019-01-16:
I was able to reverse engineer the commands. I'm leaving them here just for searchability if someone else has this issue. I've integrated it in to my Arduino `Makefile` when you call `make spiffs`

[Makefile](https://www.perturb.org/code/Arduino-Makefile)

### 1MB Spiffs
```
mkspiffs -c ~/Arduino/ds18b20/data/ -p 256 -b 8192 -s 1028096 /tmp/out.spiffs
esptool -cd nodemcu -cb 460800 -cp /dev/ttyUSB0 -ca 0x300000 -cf /tmp/out.spiffs
```

### 2MB Spiffs
```
mkspiffs -c ~/Arduino/ds18b20/data/ -p 256 -b 8192 -s 2076672 /tmp/out.spiffs
esptool -cd nodemcu -cb 460800 -cp /dev/ttyUSB0 -ca 0x200000 -cf /tmp/out.spiffs
```

### 3MB Spiffs
```
mkspiffs -c ~/Arduino/ds18b20/data/ -p 256 -b 8192 -s 3125248 /tmp/out.spiffs
esptool -cd nodemcu -cb 460800 -cp /dev/ttyUSB0 -ca 0x100000 -cf /tmp/out.spiffs
```

closed 2019-01-16 by scottchiefbaker

matthijskooijman, 2019-01-16:
Thanks for the followup, good to hear you sorted things out :-)