arduino/Arduino #8392: Pro Mini baud rate error

Parms01, 2019-01-14:
IDE 1.8.8 ... Serial.begin(115200) actually establishes serial comms at 57600.

per1234, 2019-01-15:
This issue is caused by you having one of the 16 MHz options selected from the **Tools > Processor** menu when your Pro Mini is actually running at 8 MHz. It can be resolved by choosing the correct **Tools > Processor** menu selection and then uploading the sketch to your Pro Mini again.

closed 2019-01-15 by per1234

Parms01, 2019-01-15:
Oops! ... Thanks.

Paul Armstrong

Cell: +  1    973-908-1925

Disclaimer and Confidentiality Note. This e-mail, its attachments and any
rights attaching hereto are, unless explicitly indicated otherwise, are the
property of the sender. The content of this e-mail is confidential, private
and intended for the addressee only. If you are not the addressee and have
received this e-mail, kindly notify the sender, delete this e-mail
immediately and do not disclose or use the same in any manner whatsoever.
The sender accepts no liability whatsoever for any loss or damages
whatsoever and howsoever incurred, or suffered, resulting, or arising, from
the use of this email or its attachments. The sender does not warrant the
integrity of this e-mail nor that it is free of errors, viruses,
interception or interference.