arduino/Arduino #8393: Adruino IDE bug in "Replace with" property

UygarDemir, 2019-01-15:
I tried to replace "adr10" to "adres[10]". But the result was "adres[1]0".

per1234, 2019-01-15:
Which version of the Arduino IDE are you using?

Which OS are you using?

UygarDemir, 2019-01-16:
I'm using Arduino 1.8.5. My computer's operating system is Ubuntu 18.10 64-bit.

Notice: I have got "adres[1]" in my code.

facchinm, 2019-01-16:
I can't reproduce it either, Linux 64 on hourly (1.8.9). Find/replace has not been updated since  before 1.8.5 so if it got fixed it is due to another unrelated bugfix
https://github.com/arduino/Arduino/commits/master/app/src/cc/arduino/view/findreplace/FindReplace.java

UygarDemir, 2019-01-16:
Soon ago I tested it again with same version. I used a simple code wich only contain "adr10". The result was normal: "adres[10]". But when I try it with my code, the result was wrong. If you need, I can send the code with e-mail. 

But when I tried with the hourly build, everything was normal.

per1234, 2019-01-17:
Since the problem doesn't occur in the hourly build, I'll go ahead and close this. 

However, you are welcome to post your code here. If it's a single file, go ahead and paste it right into a reply. If it's multiple files, you can put them in a .zip file and attach them. Thanks.

The code could be very helpful in case we need to investigate this issue later since it sounds like specific conditions must be in place and we need to be able to reproduce the bug. Since there were no changes to the relevant code between 1.8.5 and the Hourly Build, it's possible the issue could still occur under the right conditions.

closed 2019-01-17 by per1234