arduino/Arduino #8394: Compiler issue

alexz1999, 2019-01-15:
I started to have compiling issues after updating couple of my boards. So, I upload a sketch and it says "Exit status -1073741819. Error compiling for board ATmega16"(I'll use ATmega16 for demonstration purpose) and I have to press upload from two to four times to get the sketch to finally compile and upload.

It occurs every time I switch boards like Arduino uno, esp8266, atmega16, etc., when I restart my PC, and when I change board stats like Flash size. Why am I getting this issue?

per1234, 2019-01-15:
Please do this:
1. File > Preferences > Show verbose output during: > compilation (check)  > OK
1. Sketch > Verify/Compile
1. After the compilation fails you'll see a button on the right side of the orange bar "Copy error messages". Click that button.
1. Paste the error messages in a reply here [using code fencing](https://guides.github.com/features/mastering-markdown/).

alexz1999, 2019-01-16:
`Arduino: 1.8.8 (Windows Store 1.8.19.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\sasha\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\sasha\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\sasha\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=enabled,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10808 -build-path C:\Users\sasha\AppData\Local\Temp\arduino_build_17441 -warnings=none -build-cache C:\Users\sasha\AppData\Local\Temp\arduino_cache_373247 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool-2.5.0-3-20ed2b9.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -verbose C:\Users\sasha\AppData\Local\Temp\untitled685422918.tmp\sketch_jan15a\sketch_jan15a.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\sasha\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\sasha\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\sasha\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=enabled,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10808 -build-path C:\Users\sasha\AppData\Local\Temp\arduino_build_17441 -warnings=none -build-cache C:\Users\sasha\AppData\Local\Temp\arduino_cache_373247 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool-2.5.0-3-20ed2b9.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\sasha\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -verbose C:\Users\sasha\AppData\Local\Temp\untitled685422918.tmp\sketch_jan15a\sketch_jan15a.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\sasha\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2
Using core 'esp8266' from platform in folder: C:\Users\sasha\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2
Detecting libraries used...
"C:\\Users\\sasha\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\sasha\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.0-beta2/tools/sdk/include" "-IC:\\Users\\sasha\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.0-beta2/tools/sdk/lwip2/include" "-IC:\\Users\\sasha\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\sasha\\AppData\\Local\\Temp\\arduino_build_17441/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fexceptions -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10808 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DESP8266 "-IC:\\Users\\sasha\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.0-beta2\\cores\\esp8266" "-IC:\\Users\\sasha\\Documents\\ArduinoData\\packages\\esp8266\\ [... cut ...] inoData\\packages\\esp8266\\hardware\\esp8266\\2.5.0-beta2\\cores\\esp8266\\core_esp8266_sigma_delta.c" -o "C:\\Users\\sasha\\AppData\\Local\\Temp\\arduino_build_17441\\core\\core_esp8266_sigma_delta.c.o"
Using previously compiled file: C:\Users\sasha\AppData\Local\Temp\arduino_build_17441\core\spiffs\spiffs_check.c.o
Using previously compiled file: C:\Users\sasha\AppData\Local\Temp\arduino_build_17441\core\uart.c.o
exit status -1073740791
Error compiling for board NodeMCU 1.0 (ESP-12E Module).`