arduino/Arduino #8396: Arduino Framework makes nonsense with timer register and serial

mikrocoder, 2019-01-16:
Arduino Framework makes nonsense.

If I do not set register TCCR2A to 0, the timer will not be set as desired.
The timer clocks incorrectly and the serial interface is disabled.

Nullt man das Register TCCR2A vorher funktioniert alles wie gewünscht.
Setzt man das Register OCR2A am Ende funktioniert es auch wie gewünscht.

Arduino Mega2560 with IDE version 1.8.8

[code]
const byte pin_T2 = 10;

void setup(void) {
  Serial.begin(9600);
  pinMode(pin_T2, OUTPUT);

  Serial.print("OCR2A");  Serial.print('\t'); Serial.println(OCR2A);    
  Serial.print("OCR2B");  Serial.print('\t'); Serial.println(OCR2B);
  Serial.print("TCCR2A"); Serial.print('\t'); formatiere_Byte (TCCR2A);
  Serial.print("TCCR2B"); Serial.print('\t'); formatiere_Byte (TCCR2B); 
  Serial.print("TIMSK2"); Serial.print('\t'); formatiere_Byte (TIMSK2);  
  Serial.flush();
  
  set_Timer2();   
 
  Serial.println();
  Serial.print("OCR2A");  Serial.print('\t'); Serial.println(OCR2A);    
  Serial.print("OCR2B");  Serial.print('\t'); Serial.println(OCR2B);
  Serial.print("TCCR2A"); Serial.print('\t'); formatiere_Byte (TCCR2A);
  Serial.print("TCCR2B"); Serial.print('\t'); formatiere_Byte (TCCR2B); 
  Serial.print("TIMSK2"); Serial.print('\t'); formatiere_Byte (TIMSK2);
  
}

void loop(void) {
  
}

    
void set_Timer2()         // CTC, Mode 2
{
  // this works not
  cli();              
  TCNT2 = 0;                       
  //TCCR2A = 0;           // Why zeros necessary?  <<< <<<
  OCR2A = 249;           
  TIMSK2 = (1<<OCIE2A);   // Compare Match A
  TCCR2A = (1<<WGM21);    // set CTC Mode
  TCCR2B = (1<<CS22);     // Prescaler 64
  sei();  

  /* this works  
  cli();              
  TCNT2 = 0;                       
  TCCR2A = 0;             // Why zeros necessary?  <<< <<<
  OCR2A = 249;           
  TIMSK2 = (1<<OCIE2A);   // Compare Match A
  TCCR2A = (1<<WGM21);    // set CTC Mode
  TCCR2B = (1<<CS22);     // Prescaler 64
  sei(); 
  */  

  /* this works  
  cli();              
  TCNT2 = 0;                               
  TIMSK2 = (1<<OCIE2A);   // Compare Match A
  TCCR2A = (1<<WGM21);    // set CTC Mode
  TCCR2B = (1<<CS22);     // Prescaler 64
  OCR2A = 249; 
  sei(); 
  */            
}  

ISR(TIMER2_COMPA_vect)          // Timer 2 ISR, normal 500Hz
{  
  static bool state = LOW;
  state = !state;
  digitalWrite(pin_T2, state);                    
}

void formatiere_Byte (unsigned int data)
{
  Serial.print(F("data: "));
  for (char i=7;i>=4;i--) {
    Serial.print( (data >> i) & 0x01);
  }
  Serial.print("'");
  for (char i=3;i>=0;i--) {
    Serial.print( (data >> i) & 0x01);
  }
  Serial.println();
}
[Code]

mikrocoder, 2019-01-16:
``const byte pin_T2 = 10;

void setup(void) {
  Serial.begin(9600);
  pinMode(pin_T2, OUTPUT);

  Serial.print("OCR2A");  Serial.print('\t'); Serial.println(OCR2A);    
  Serial.print("OCR2B");  Serial.print('\t'); Serial.println(OCR2B);
  Serial.print("TCCR2A"); Serial.print('\t'); formatiere_Byte (TCCR2A);
  Serial.print("TCCR2B"); Serial.print('\t'); formatiere_Byte (TCCR2B); 
  Serial.print("TIMSK2"); Serial.print('\t'); formatiere_Byte (TIMSK2);  
  Serial.flush();
  
  set_Timer2();   
 
  Serial.println();
  Serial.print("OCR2A");  Serial.print('\t'); Serial.println(OCR2A);    
  Serial.print("OCR2B");  Serial.print('\t'); Serial.println(OCR2B);
  Serial.print("TCCR2A"); Serial.print('\t'); formatiere_Byte (TCCR2A);
  Serial.print("TCCR2B"); Serial.print('\t'); formatiere_Byte (TCCR2B); 
  Serial.print("TIMSK2"); Serial.print('\t'); formatiere_Byte (TIMSK2);
  
}

void loop(void) {
  
}

    
void set_Timer2()         // CTC, Mode 2
{
  // this works not
  cli();              
  TCNT2 = 0;                       
  //TCCR2A = 0;           // Why zeros necessary?  <<< <<<
  OCR2A = 249;           
  TIMSK2 = (1<<OCIE2A);   // Compare Match A
  TCCR2A = (1<<WGM21);    // set CTC Mode
  TCCR2B = (1<<CS22);     // Prescaler 64
  sei();  

  /* this works  
  cli();              
  TCNT2 = 0;                       
  TCCR2A = 0;             // Why zeros necessary?  <<< <<<
  OCR2A = 249;           
  TIMSK2 = (1<<OCIE2A);   // Compare Match A
  TCCR2A = (1<<WGM21);    // set CTC Mode
  TCCR2B = (1<<CS22);     // Prescaler 64
  sei(); 
  */  

  /* this works  
  cli();              
  TCNT2 = 0;                               
  TIMSK2 = (1<<OCIE2A);   // Compare Match A
  TCCR2A = (1<<WGM21);    // set CTC Mode
  TCCR2B = (1<<CS22);     // Prescaler 64
  OCR2A = 249; 
  sei(); 
  */            
}  

ISR(TIMER2_COMPA_vect)          // Timer 2 ISR, normal 500Hz
{  
  static bool state = LOW;
  state = !state;
  digitalWrite(pin_T2, state);                    
}

void formatiere_Byte (unsigned int data)
{
  Serial.print(F("data: "));
  for (char i=7;i>=4;i--) {
    Serial.print( (data >> i) & 0x01);
  }
  Serial.print("'");
  for (char i=3;i>=0;i--) {
    Serial.print( (data >> i) & 0x01);
  }
  Serial.println();
}

PaulStoffregen, 2019-01-16:
You should ask for help on the forum.

closed 2019-01-16 by facchinm