arduino/Arduino #8397: ctags component needs updating to 64 bit for MacOS

dap35, 2019-01-16:
Arduino v1.8.8:
This is listed under legacy 32bit code in MacOS.  This needs to get updated prior to MacOS v10.15, which reportedly removes support for 32bit legacy mode.

ctags:

  Last Used:	2018-12-22 10:43:17
  Path:	/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags
  Parent Process:	/Applications/Arduino.app/Contents/MacOS/Arduino

SConaway, 2019-02-26:
Where is that coming from?

dap35, 2019-02-26:
[image]

In MacOS, go to the System Information tool, > Software > Legacy Software.

SConaway, 2019-03-03:
Yes. Why is this in the app?

dap35, 2019-03-03:
I have no clue as to why it is there.  I simply installed the application and wanted to report this so that it can be dealt with proactively.

facchinm, 2019-03-04:
`ctags` is being used for prototype generation. It is going to be replaced by `arduino-preprocessor` in the near future, so I'd avoid recompiling it now

Mike4U, 2019-03-14:
I'm getting warnings about other processes on MacOS 10.14.3 (18D109) in Arduino 1.8.8:
[image]

These are all in: /Applications/Arduino.app/Contents/Java/hardware/tools/avr/

ttimpe, 2019-06-04:
This needs fixing ASAP as the developer preview of macOS Catalina(10.15) now doesn't run any 32-bit binaries anymore.

SConaway, 2019-06-23:
What's going on here? Arduino's the only thing keeping me on Mojave.

@facchinm @per1234

ttimpe, 2019-06-23:
This seriously needs to be addressed very soon. It should just be a small recompile, many (if not all) tools already compile just fine.

SConaway, 2019-06-23:
It seems to build just fine on my Mac.

facchinm, 2019-06-25:
Everything was rebuilt to 64 bit and the resulting IDE build is here https://github.com/arduino/Arduino/pull/8976 .
Can you test it and report if everything is working fine?

SConaway, 2019-06-25:
Unfortunately, I'm not home for the rest of the week, but as soon as I return, I'll test.

Mike4U, 2019-06-25:
Used the link on 10.14.5 and got these 32bit.  Strangely worse than Arduino 1.8.7 which only had ctags
[image]

facchinm, 2019-06-26:
@Mike4U pretty sure you are not looking at build #8976 since it ships with avr-gcc 7.3.0 and your screenshot shows 5.4.0 :wink:

Mike4U, 2019-06-26:
@facchinm  Not sure what the About this Mac/System report above is actually doing.  If I click on the 'Applications' on the left Arduino will show up as 64 bit.  The 'Activity Monitor' application also shows it as 64 bit when it's running.  The 'Legacy Software' clicked above shows the same application (at the same location) with 32bit components inside it.

This isn't causing me any problems AFAIK, just posted for your info.

Mike4U, 2019-06-26:
@facchinm I have older Arduino1.6.6 and 1.8.9 in a folder with-in the Applications folder.  The 8976-BUILD-869 does indeed have the air-gcc 7.3.0, not 5.4.0 as you indicated.  However the 'About this Mac->System report->Legacy Software, incorrectly shows the 1.8.9 version in the Application folder not Application/old version/ where it is.  Perhaps it's not updating from the menu.  I'll try a restart.

Mike4U, 2019-06-26:
Restart didn't help.  I give up on debugging Apples App.  Arduino is fine.

SConaway, 2019-06-30:
The menu doesn't update unless the offending files are removed. (I think)

sandeepmistry, 2019-09-16:
@facchinm I think `ctags` is still 32-bit in the latest IDE download for macOS.

facchinm, 2019-09-17:
I just downloaded 1.8.10 and can confirm it's 64 bit :wink: 
```
$ file ctags 
ctags: Mach-O 64-bit x86_64 executable, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL|PIE>
```

closed 2019-09-17 by facchinm

sandeepmistry, 2019-09-17:
Strange, the UI shows this:

[image]

... but let's trust the terminal :)