arduino/Arduino #8398: No serial port for arduino uno r3 on mac OS X MOJAVE when plugged

lukassn, 2019-01-17:
I've tried so many things to fix this problem, but nothing worked. I have an Arduino UNO R3 and macbook with OS X Mojave (10.14). There's no usb serial port available when I plug the Arduino on machine. Could you help me, please? I'm trying to teach (and learn) with my little brother about engineering.

per1234, 2019-01-17:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. If you're using an Uno derivative with the CH340 USB to TTL serial adapter chip (usually what you'll get if you buy a board on eBay, Aliexpress, etc.), then read this thread first, as it might give you the solution: https://forum.arduino.cc/index.php?topic=570440.0. If that doesn't apply to you, then proceed to the next step:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to provide the information as we'll need to help.

closed 2019-01-17 by per1234

cyruslk, 2020-09-05:
Have you been able to sold the issue? I have the same issue

lukassn, 2020-09-07:
@cyruslk from what I remember, I was using an unofficial arduino. I bought an official model and it worked!