arduino/Arduino #8401: [Library Manager] Please add new library

Heltec-Aaron-Lee, 2019-01-18:
Please add the following to the library list:
[https://github.com/HelTecAutomation/Heltec_ESP32](https://github.com/HelTecAutomation/Heltec_ESP32)

Thanks.

Aaron Lee

Heltec-Aaron-Lee, 2019-01-20:
@per1234 Thanks for your add.

But I still can't find my library in Arduino Library manager, what can I do now?
[image]

per1234, 2019-01-20:
I only labeled the issue to make it easy to find in the issue tracker. Someone else is in charge of adding it to the Library Manager. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

Heltec-Aaron-Lee, 2019-01-21:
@per1234 Thank you very for your detail reply. Have a good day.

cmaglie, 2019-01-21:
done

closed 2019-01-21 by cmaglie