arduino/Arduino #8402: 1.9.0-beta-linux64:  Setting UDEV

richard9wirthware, 2019-01-18:


richard9wirthware, 2019-01-18:
In arduino-linux-setup.sh
In line 91 and 92 there a syntax error. "KERNEL="ttyUSB[0-9]*" my  system complains about...
It should be:  Kernel=="ttyUSB[0-9]*"  and so on

closed 2019-01-18 by richard9wirthware

luco5826, 2019-05-30:
Hi!
I think this could be related: I found in my `journalctl` log these two lines
```
mag 30 17:04:57 debian systemd-udevd[316]: /etc/udev/rules.d/90-extraacl.rules:4: Unknown key '"KERNEL'
mag 30 17:04:57 debian systemd-udevd[316]: /etc/udev/rules.d/90-extraacl.rules:5: Unknown key '"KERNEL'
```
After checking the file it was right: rules in `extraacl.rules` file start with a " character. 
Is it intended or not?
I didn't experienced any problem until now but I thought this could be an error!