arduino/Arduino #8405: [Library Manager] Add Antares ESP8266 HTTP and Antares ESP8266 MQTT to Library Manager

vmasdani, 2019-01-19:
Greetings.
Please add Antares ESP8266 HTTP and also Antares ESP8266 MQTT to the Library Manager, thank you.
https://github.com/antaresdocumentation/antares-esp8266-http
https://github.com/vmasdani/antares-esp8266-mqtt

per1234, 2019-01-19:
There are a few issues that need to be resolved before your libraries can be added to the Library Manager index:

#### Antares ESP8266 HTTP
- You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

#### Antares ESP8266 MQTT
- You need to fill out the [url field in the library.properties file](https://github.com/vmasdani/antares-esp8266-mqtt/blob/master/library.properties#L8). It is currently blank.
- You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

---
After resolving these issues, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

vmasdani, 2019-01-19:
> There are a few issues that need to be resolved before your libraries can be added to the Library Manager index:
>
> #### Antares ESP8266 HTTP
> * You need to create a [...]

Thank you for the quick response. I have fixed the `url` field on  `library.properties` and also added a release on both libraries.

closed 2019-01-21 by cmaglie