arduino/Arduino #8408: In the last two years we updated `jmdns` (which is being used to discover network boards) to address security and performance problems in the older versions. The boards are now added/removed based on the information coming from that library, so if a board disappears it is due to mdns reporting the "disconnection".

jcflyer, 2019-01-20:
I have used Arduino IDE 1.6.11 on a wired desktop running XP SP2 to wirelessly program 4 Nodemcu devices without any problem for over a year.
I got a new laptop with Windows 10 and installed 1.6.11 and all the network ports show up and work immediately but then disappear and the port is greyed out in a minute or so. 
I can only upload once or twice before the port is gone.
If I restart the laptop the ports show up and work again for a short time. Nothing I have tried has worked so far including allowing port 8266 or turning off the firewall.
Also compiling is very slow now.

per1234, 2019-01-20:
It's not useful to report problems that occur with outdated versions of the Arduino IDE. Please always download the Hourly Build of the Arduino IDE and test with that version to verify the problem still occurs before opening an issue report:
https://www.arduino.cc/en/Main/Software#hourly

After doing so, please comment here to let us know whether your issue has been resolved or not.

jcflyer, 2019-01-20:
OTA only works with old ide versions. New version does the same thing.

facchinm, 2019-01-21:
@jcflyer in the original post you are talking about using 1.6.11 in both XP and windows10 and mdns discovery only works in the former, right? Or are you comparing 1.6.11 with 1.8.8?

jcflyer, 2019-01-21:
Thanks for your reply.

Correct. I have used 1.6.11 on XP for over a year with no issues. At the
time newer versions wouldn't work with Nodemcu boards. I also use old
Python 2.7.13 as required for OTA uploading at the time.

It's weird that on Windows 10 1.6.11 works fine, finds all the boards and
uploads OK but only for a minute or so. Then the port is greyed out and you
have to restart to try again. All of the boards are still on the network as
they all host web pages that still work and they all still show up on XP.

Something happens after a minute or so on Windows 10 that kills the ports
showing up. XP with wired ethernet on a desktop is no problem.

John

jcflyer, 2019-01-21:
Attached are screenshots from Windows 10 showing all ports initially then
Port greyed out after a minute. XP doesn't do this on wired ethernet. My
laptop doesn't have an ethernet port so I can't try it wired.

facchinm, 2019-01-23:
Hi @jcflyer ,
can I ask you to test the latest Beta https://www.arduino.cc/en/Main/Software#beta ?
You'll see two menus for both serial ports and mdns ports; one is bundled while the other is pluggable and based on https://github.com/arduino/mdns-discovery . 
If the pluggable one works better it would greatly accelerate the switch.
Let me know! Thanks!

jcflyer, 2019-01-23:
The problem was the Python version I was using with Windows 10. Python
3.7.2 works OK but 2.7.13 makes the ports disappear after a minute on
Windows 10. 2.7.13 is fine with XP.

I followed instructions for OTA programming and it said you had to use old
versions of Arduino IDE and old Python version 2.7.13 or it wouldn't work.

Normally all my programming was done with an old XP SP2 desktop, Arduino
1.6.11 and the web page HMIs hosted on the Nodemcus were displayed and
interacted with using my Android phone, tablet or tv box.

All the home automation programs are uploaded wirelessly OTA since the
Nodemcus are remotely located. All of this has worked great for over a year.

All the web pages display quickly in Android Chrome but painfully slowly in
XP or Windows 10.

If you want to see the web pages and what I am doing go to:
https://sites.google.com/site/nodemcu12e/

Code for one of them is a the bottom. It's a  mishmash of code cut and
pasted from many sources. It works fine with XP and Android.

My Android tablet died so I am trying to replace it with a new Windows 10
laptop. I want to use it to program instead of XP and display the web page
HMIs as well. Neither OTA programming nor web page display works properly
in my case.

The Python update someone suggested fixed the ports disappearing problem
but uploading OTA only works with very small programs like Blink and none
of the web pages display properly or at all. The Nodemcus often crash
requiring a reload of Basic OTA if I try to load the identical working
programs from Windows 10 instead of XP.

I tried Arduino 1.8.8 and Python 3.7.2 but I have the same problems. Only
Arduino 1.6.11 on XP with Python 2.7.13 for OTA uploading works properly in
my case. The programs are fairly large since they include a lot of html
code for the web pages.

Web page display on Android Chrome devices takes less than 2 secs. Same web
page display on Windows 10 desktop Chrome is 22+ secs or won't display at
all. Win10 with Internet Explorer browser is 5 secs or so but one web page
won't display at all.

I have had nothing but problems for days now trying to get Windows 10
working the same as XP SP2 and Android Chrome for web page display that has
worked properly for over a year now. Very frustrating.

John

Bbas11, 2019-01-24:
I had the same problem with ports disappearing  since 1.8.6. If i restart the IDE they would show for a few seconds then there gone. Latest beta fixed it. Now two ports show for each board. if i select one the other one gets selected also. All the ports always show up now in both versions now. No problems with flashing. Looks like its finally fixed. 
Windows 10, python 2.7 basic arduinoOTA sketch.

pfeerick, 2019-01-24:
I think the issue is in the Arduino IDE, as when I open the ports menu the first time after opening the IDE (v1.8.8) on linux, I see four of the five mDNS OTA ready devices listed... if I close the menu (not the IDE), and immediately reopen the ports menu... all but one is missing (and which one is shown is sometimes random between restarts of the Arduino IDE). But I can see all five devices in the PlatformIO mDNS list, and also using zeroconfbrowser - so they are reporting correctly, but the IDE does not **show** them. I have the same issue of not all devices showing on Windows (also v1.8.8), but I can't swear that it exhibits the same "some then one" behaviour. I tried running 1.8.1 of the IDE on Windows just after 1.8.8, and all of the devices are listed.

@facchinm I'll try the beta version on Windows also, and let you know what my results are ;)

facchinm, 2019-02-07:
@pfeerick board menu duplication is intended (one is the internal java discoverer, the other list is the pluggable one from #8038). If the list diverge, we have a bug :stuck_out_tongue:

immutability, 2019-05-13:
I don't want to open a new issue report in case it would be a duplicate, as I'm possibly facing the same problem on my Windows 10 machine. In previous Arduino IDE versions (tested on 1.8.3 and 1.8.5) the port discovery works just fine and is reliable and stable. With versions 1.8.6, 1.8.7, 1.8.8, 1.8.9 and the current 1.9.0 beta, the port discovery doesn't work at all for me - I only have one ESP8266 node on my network but it never shows up. Switching between different Python versions doesn't seem to have any effect, currently on Python 3.7.2.

Other mDNS browsers find the ESP8266 node without a problem (tried a Chrome app and Bonjour Browser from hobbyistsoftware). Seeing the reference to jmdns library, I have tried to run the sample discovery java code provided in jmdns readme on github, and it doesn't find anything on my Windows 10 machine. Running the same java sample code on Xubuntu 18.04 VM on the same network works OK and discovers the ESP8266 node. Unsurprisingly, Arduino IDE 1.8.9 also finds it on Xubuntu 18.04.

Do I understand the above note right that it will soon be possible to hard-code known ports that aren't detected automatically using a text file?

Bbas11, 2019-05-13:
Try the hourly. The only problem i still have is once my computer sleeps the ports wont come back unless i restart. Otherwise the ports are always there.
Tested with Windows 10 pro, python 2.7, basic arduinoOTA sketch.

immutability, 2019-05-14:
@Bbas11 thanks for the suggestion, haven't tried the hourly builds yet so I did now - Arduino 1.8.10 hourly build 2019/05/07 05:33 - but alas, no change, nothing detected. I have also tried a clean boot as I'm using sleep all the time and rarely reboot but it made no difference.

hbosko, 2019-05-20:
i fixed my issue with:
First start arduino 1.8.9 - no ports are showing even if i wait 100 years... dont close that one...
in virtual mashine or the same PC i started version 1.8.5 and immediately are ports showing on both arduino versions.

If i restart arduino 1.8.9 and ver 1.8.5 is running in VM or background, no ports will show immediately, but will in about 1 minute.

ESP8266 version 2.5.2. - both arduino versions
Phyton 2.7.15 64bit

nothing blocked in firewall, python, java or arduino.

You NEED to run version 1.8.5 on the SAME PC.

Can you please try if works for you?

immutability, 2019-05-21:
@hbosko wow, this sounds like such a crazy idea that I had to try it :) and it worked! I haven't tried it in a VM but I started 1.8.9 first, and then 1.8.5 and voila, the network port appeared in both running instances of the Arduino IDE. 

If I started 1.8.9 while 1.8.5 was already running, the port only appeared in 1.8.9 after about 3 or 4 minutes (so a bit longer than for you) but when I first start 1.8.9 and only then 1.8.5, the port will appear in both pretty much immediately. **Interesting workaround - thanks**! :)

What is quite interesting is that if I launch Bonjour Browser while Arduino 1.8.9 is running, it seems to have the same effect - as soon as the device appears in the Bonjour Browser, the port also appears in Arduino 1.8.9. So another workaround without the need to keep multiple versions of Arduino IDE installed/running side by side.

I did leave the Arduino IDE 1.8.9 running for 20 minutes on its own now (with 1.8.5 and the Bonjour Browser closed) just to see if the port wouldn't eventually appear after a longer delay, but no, it didn't. After starting the Bonjour Browser (after the 20 minutes) the port appeared pretty much immediately.

hbosko, 2019-05-21:
Glad it worked and you found easyier way to find ports without running 2 instances, since i keep my computer clean as posible, without any additional software... 
So meybe we are closer to be fixed :)

This is my simple OTA code, it works fantastic to the version 1.8.5<, not a single problem. As you see i removed mDNS and didnt had any trouble on host maschines winXP or win7 or win10, not tried linux.

`
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid =      "InterneT";
const char* password =  "Free4ALL";
const char* hstname =   "ESP8266_OTA";

#define led 2

char hostString[16] = {0};

void setup() {
  pinMode(led, OUTPUT);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  sprintf(hostString, hstname, ESP.getChipId());
  WiFi.hostname(hostString);
  unsigned long connect_start = millis();
  while(WiFi.status() != WL_CONNECTED) {
    delay(500); Serial.print(".");
    if(millis() - connect_start > 30000) {
      ESP.reset();}
  }  
  ArduinoOTA.setHostname(hstname);
  // ArduinoOTA.setPassword((const char *)"123");
  ArduinoOTA.begin();
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();

  digitalWrite(led, HIGH);
  delay(500);
  digitalWrite(led, LOW);
  delay(1000);
  
}

``

I am running PFsense 2.4.4. router - i blamed router but i was wrong.

Meybe DEVs removed smthing smthing, i am not programmer so i cant look code and find solution, but i hope it hepled litlebit :)

Sorry for my bad english.

FunnyAnimal, 2020-06-02:
Please upgrade 'jmdns' library to 3.5.5 (or later).

Arduino IDE versions 1.8.6 ... 1.8.12 are having problems showing devices as network ports for OTA upgrade.

'jmdns' version 3.5.3 has problems on systems with multiple network interfaces. (By the way: version 3.5.1, as used by IDE 1.8.5 and below, works fine...)

See analysis @ https://forum.arduino.cc/index.php?topic=575560.msg4608533#msg4608533 .

per1234, 2022-10-11:
Closing as obsoleted by the complete reworking of the network port discovery system for Arduino IDE 2.0.0.

If anyone is having problems with network ports on Arduino IDE 1.x, please give 2.x a try. The download links are listed here:

https://www.arduino.cc/en/software/

If you discover any problems, feel free to open an issue in the new repository:

https://github.com/arduino/arduino-ide/issues?q=

closed 2022-10-11 by per1234