arduino/Arduino #8412: Arduino battery class

offcircuit, 2019-01-20:
My libs

https://github.com/offcircuit/Battery
https://github.com/offcircuit/DTime

per1234, 2019-01-20:
In order to be added to the Library Manager index, your libraries must contain a library.properties file. The format of this file is documented here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) in each repository that matches the `version` value in your library.properties file.

After resolving these issues, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

offcircuit, 2019-01-20:
Hi again,

I just complete your requests, on both of the libs.

Thanks,
Helder Rodrigues

per1234, 2019-01-20:
DTime looks good but Battery still has an issue:

Each library in the Library Manager index is required to have a unique name (as defined by the `name` field in library.properties). Battery's [`name` field](https://github.com/offcircuit/Battery/blob/1.0/library.properties#L1) is currently set to "Time" but there is already a library of that name in the Library Manager index.

Please change the `name` value to something unique and then redo the release/tag.

After resolving these issues, please comment here to let us know we can proceed with adding your library to the Library Manager index.

offcircuit, 2019-01-20:
Hi again,

I changed to Batflow.
with new url: https://github.com/offcircuit/Batflow

Thanks a lot
Helder Rodrigues

cmaglie, 2019-01-21:
done

closed 2019-01-21 by cmaglie