arduino/Arduino #8414: Error https://www.arduino.cc/reference/pt/language/variables/data-types/string/

ckwiek, 2019-01-21:
The link https://www.arduino.cc/reference/pt/language/variables/data-types/string/ in the example have a esintaxe error, the text:
char* mStrings[]={"This is string 1", "This is string 2", "This is string 3",
"This is string 4", "This is string 5","This is string 6"}; produce an compilation error because the correct is:

char myString [ ] and nor mString[ ] as posted in the page.

Thank you;
Best regards.
Carlos.

facchinm, 2019-01-21:
Hi @ckwiek ,
can you move the issue to https://github.com/arduino/reference-pt ? Thanks!

closed 2019-01-21 by facchinm

per1234, 2019-01-26:
@ckwiek I've submitted a pull request (https://github.com/arduino/reference-pt/pull/232) to fix this error so there is no need for you to open the issue in the reference-pt repository now.

ckwiek, 2019-01-26:
> @ckwiek I've submitted a pull request (arduino/reference-pt#232) to fix this error so there is no need for you to open the issue in the reference-pt repository now.

Thank you!

Regards,

Desculpe a brevidade, enviado pelo meu IPhone - Perdone Los errores de tipeo.

Att, 

Carlos Kwiek