arduino/Arduino #8428: [Library Manager] Please add the RTCCounter and CryptoAuthLib libraries to the Library Manager

GabrielNotman, 2019-01-23:
Can you please add these two libraries (RTCCounter and CryptoAuthLib) to the Library Manager:

- https://github.com/GabrielNotman/RTCCounter
- https://github.com/GabrielNotman/cryptoauthlib

Thanks

per1234, 2019-01-23:
All libraries in the Library Manager index are required to have their library.properties `url` field defined. Yours are currently blank:
- https://github.com/GabrielNotman/RTCCounter/blob/1.0.0/library.properties#L8
- https://github.com/GabrielNotman/cryptoauthlib/blob/1.0.0/library.properties

Please resolve these issues and then do a tag/release.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

GabrielNotman, 2019-01-23:
@per1234 Thanks, I've have added that field now.

For both libraries Tag/Release 1.0.0 has that change included.

GabrielNotman, 2019-01-23:
One other thing, the cryptoauthlib library is a fork of the main repository from MicroChip. They have a 11 older tags/releases which uses a date based system (e.g. 20190104). Will these cause issues with the Library Manager?

per1234, 2019-01-23:
Looks good.

I don't think the old releases will be a problem. Library Manager will not index any release that doesn't meet the requirements so they will not be available for installation, but your compliant 1.0.0 release will.

cmaglie, 2019-01-29:
done!

closed 2019-01-29 by cmaglie