arduino/Arduino #8433: [Library Manager] Add Tinker Controller and SensorToButton

nathanRamaNoodles, 2019-01-24:
Tinker Controller: https://github.com/nathanRamaNoodles/TinkerController-Library

SensorToButton: https://github.com/nathanRamaNoodles/SensorToButton

per1234, 2019-01-24:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

nathanRamaNoodles, 2019-01-24:
Ok, I've created releases for both Libraries. You may proceed.

nathanRamaNoodles, 2019-01-28:
It's been 4 days, and the libraries are not visible in Library Manager.  Could you try again?

per1234, 2019-01-28:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

cmaglie, 2019-01-29:
done!

closed 2019-01-29 by cmaglie