arduino/Arduino #8435: Library manager not showing latest library.properties

nseidle, 2019-01-24:
[This library](https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library) is displaying oddly in lib manager.

[image]

Library name and library description are wrong. Link is correct. Version drop down is not showing a few of the latest additions

I changed the library.properties version before doing a release. And I don't see any [symlinks in the repo](https://github.com/arduino/Arduino/issues/8061) that would cause it not to be picked up.

Library: https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library

Original request for inclusion: https://github.com/arduino/Arduino/issues/6225

I suspect I'm doing something wrong with our repo but I can't spot it. Sorry to bother.

facchinm, 2019-01-25:
Hi @nseidle ,
the library engine was looking for the original name of the library (SparkFun CCS811 Breakout). I changed it to the new name and newer versions should be available soon!

closed 2019-01-25 by facchinm

nseidle, 2019-01-25:
Fixed! Thank you!