arduino/Arduino #8449: Please add the new lib ModbusConfig

Warlib1975, 2019-01-27:
Library located at https://github.com/Warlib1975/ModbusConfig.git

ModbusConfig library for ESP8266/ESP32/Arduino.
The library process a Modbus configuration file in JSON format.

per1234, 2019-01-27:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Warlib1975, 2019-01-28:
> You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file. [...]

Done. :-)

cmaglie, 2019-01-29:
done

closed 2019-01-29 by cmaglie