arduino/Arduino #8457: Incorrect error line highlight when error is inside folded code

per1234, 2019-01-28:
Using Arduino IDE 1.8.9 Hourly Build 2019/01/24 07:33 and Arduino IDE 1.9.0-beta build 100 with Windows 10 64 bit

1. Create the following sketch in the Arduino IDE:
    ```c++
    void setup() {}
    void loop() {
      #error
    }
    ```
1. File > Preferences > Enable Code Folding (check) > OK
1. Click the [-] icon to the left of the line:
    ```c++
    void loop() {
    ```
1. Sketch > Verify/Compile

The first line of the sketch will be highlighted in pink, incorrectly indicating the error is there.

Suggested improvements on this behavior:
- Unfold the code section with the error and place the highlight on the correct line.
- Don't do an error highlight when it cannot be correctly placed. The user can always get the line number from the error message in the console.
- Place error highlight at the start of the fold. This is still slightly confusing, but it's more intuitive than the current behavior.
- Unfold all on error. This seems like it could be annoying to the user so it might just be trading a new issue for an old one.

I have a suspicion this is a limitation of rSyntaxTextArea rather than the Arduino IDE, but I'm not certain of it, so I thought it was worth reporting here.

Originally reported: http://forum.arduino.cc/index.php?topic=443558.msg4036044#msg4036044

facchinm, 2019-01-28:
I remember we had a similar issue in the past and the solution was to unfold all the code before highlighting the line. It was surely related with rSyntaxTextArea not behaving properly but I have to check in the past issues to make sure it's not a regression

facchinm, 2019-01-28:
Found it, it was not totally related but probably the fix is the same
https://github.com/facchinm/Arduino/commit/61828b34888b682c1f9b38806cd2b156d94a55c7

PR #8458 (2019-03-07): Ensure line is unfolded when highlighting for error
Fixes #8457

For some reason, `getCurrentTab().getTextArea().getFoldManager().ensureOffsetNotInClosedFold(line)` doesn't work here; there no documentation on what offset is.
Also, `getFoldForLine(line)` returns null even if the line is folded (bug in rsyntaxtextarea?)

@per1234
commit 9518aa42 (2019-03-07): Ensure line is unfolded when highlighting for error

Fixes #8457

For some reason, getCurrentTab().getTextArea().getFoldManager().ensureOffsetNotInClosedFold(line) doesn't work here; there no documentation on what offset is.
Also, getFoldForLine(line) returns null even if the line is folded (bug in rsyntaxtextarea?)
closed 2019-03-07 by facchinm