arduino/Arduino #8459: 1.8.8 Arduino/ArduBlock

aroedercls, 2019-01-28:
Previously on 1.8.5 we could follow the steps below and get ArduBlock to work on every user on the computer.  Now with 1.8.8, we have to follow these steps for every computer, for every user.  Any fixes?

1.	Go to https://www.arduino.cc/en/Main/Software and download the Arduino IDE installer appropriate to your Operating System 
2.	Run the installer after the download has completed 
3.	Open Arduino 
4.	Click File – Preferences 
5.	Copy https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json and paste it into the open field next to Additional Boards Manager URLs: 
6.	Click OK 
7.	Click Tools – Boards - Board Manager and do a search for Digital Sandbox 
8.	Click the Install button for 1.1.9 or newer for SpartFun AVR Boards by SparkFun Electronics 
9.	Click Close 
10.	Click File – Preferences 
11.	Under Sketchbook location: change it to say C:\Program Files (x86)\Arduino 
12.	Click OK 
13.	Close Arduino 
14.	Go to http://www.carobot.cc/how-to/ardublock/ and click the blue words that say Download it from our server 
15.	Open your downloads folder and ensure the ardublock-161216-carobot.jar (numbers may vary) is there 
16.	Open another File Explorer window, folder icon with a blue handle on it 
17.	Navigate to C:\Program Files (x86)\Arduino 
18.	Right-click and create a new folder called tools and open it 
19.	Right-click and create a new folder called ArduBlockTool and open it 
20.	Right-click and create a new folder called tool and open it 
21.	Drag the ardublock-161216-carobot.jar (numbers may vary) and drop it into the most recent tool folder you just created 
a.	The path should now look like C:\Program Files (x86)\Arduino\tools\ArduBlockTool\tool\ ardublock-161216-carobot.jar (numbers may vary) 
22.	Close your File Explorer window 
23.	Open Arduino 
24.	Click Tools and make sure you have the correct board selected under Boards 
25.	Click Tools and make sure you have the correct port selected under Ports 
26.	Click Tools and you should notice that ArduBlock is now installed 
That will get your Arduino software fully updated, all boards installed, and ArduBlock included.

per1234, 2019-01-28:
> Now with 1.8.8, we have to follow these steps for every computer, for every user.

What is the problem you encounter if you don't do that? Please provide the full text of any error messages.

> Any fixes?

If you want a single installation to work for multiple users, I would suggest that you try running the Arduino IDE in portable mode:
https://www.arduino.cc/en/Guide/PortableIDE

---
The instructions you posted should only ever work for a single user, unless the Arduino IDE is in portable mode.

---
>Under Sketchbook location: change it to say C:\Program Files (x86)\Arduino

That is not good advice. That is the usual location of the Arduino IDE installation. Everything under that folder will be lost every time you update to a new version of the Arduino IDE.

---
>Navigate to C:\Program Files (x86)\Arduino
Right-click and create a new folder called tools and open it

It would be better to do this in the sketchbook folder. The current instructions will cause you to unnecessarily need to repeat this process every time you update to a new version of the Arduino IDE. As long as you don't put the sketchbook folder inside the Arduino IDE installation folder, it will persist through IDE updates.

aroedercls, 2019-01-28:
When we follow those steps, they work just fine.  The problem is that with 1.8.8 we’re finding every user that logs onto the computer has to do those same steps.  With 1.8.5, once you completed the steps, it would work for all users.

[chrome_2018-06-11_10-49-25]
Ashley Roeder
Senior System Support Specialist

Creative Learning Systems
1801 Lefthand Circle, Unit A
Longmont, CO 80501
Preferred Phone: 800-458-2887
Direct Phone: 800-458-2880 ext. 132
aroeder@clsonline.com<mailto:aroeder@clsonline.com>
support@clsonline.com<mailto:support@clsonline.com> (Preferred)

https://www.arduino.cc/en/Guide/PortableIDE

per1234, 2019-01-28:
>The problem is that with 1.8.8 we’re finding every user that logs onto the computer has to do those same steps.

I understand that. The information I'm trying to get from you is exactly what the problem is that requires each user to repeat the steps.

>With 1.8.5, once you completed the steps, it would work for all users.

The only way that would be possible is if the Arduino IDE was running in portable mode. When not in portable mode, the preferences are saved to a user-specific folder and the SparkFun AVR Boards package is also installed to a user-specific folder. That's how it's always been.

Please try the portable mode solution and then let me know if it solved the problem.

aroedercls, 2019-01-28:
Please correct me if I’m wrong.

I downloaded the compressed version and put the extracted folder into C:\Program Files (x86)\Arduino
I created a new folder inside of that called portable
I pasted http://www.carobot.cc/how-to/ardublock/ into File > Preferences and downloaded the SparkFun AVR

If I don’t open Arduino and run as an admin, then I get an error “Could not write preferences file: C:\Program Files (x86)\Arduino\arduino-1.8.8\portable\preferences.txt (Access is denied)” If I open and run as admin, there’s no error

And where do I download and save ardublock-161216-carobot.jar for ArduBlock to show up?  Is it still in C:\Program<file:///C:/Program> Files (x86)\Arduino\tools\ArduBlockTool\tool\ ardublock-161216-carobot.jar

[chrome_2018-06-11_10-49-25]
Ashley Roeder
Senior System Support Specialist

Creative Learning Systems
1801 Lefthand Circle, Unit A
Longmont, CO 80501
Preferred Phone: 800-458-2887
Direct Phone: 800-458-2880 ext. 132
aroeder@clsonline.com<mailto:aroeder@clsonline.com>
support@clsonline.com<mailto:support@clsonline.com> (Preferred)

aroedercls, 2019-01-28:
Ok good.  That seems to work on every user for this computer now.

I have 15 other computers to get this going on.  Would I complete all those steps on each computer or can I copy and paste the portable folder to one location?

[chrome_2018-06-11_10-49-25]
Ashley Roeder
Senior System Support Specialist

Creative Learning Systems
1801 Lefthand Circle, Unit A
Longmont, CO 80501
Preferred Phone: 800-458-2887
Direct Phone: 800-458-2880 ext. 132
aroeder@clsonline.com<mailto:aroeder@clsonline.com>
support@clsonline.com<mailto:support@clsonline.com> (Preferred)

Yeah, Windows 10 is very restrictive of the C:\Program Files (x86) folder. I recommend using the "Windows ZIP file for non admin install" download from https://www.arduino.cc/en/Main/Software and then extract the .zip file to some folder other than C:\Program Files (x86).

You can do that, but I recommend installing it to {sketchbook folder}\tools\ArduBlockTool\tool\ ardublock-161216-carobot.jar, where {sketchbook folder} is configured in the Arduino IDE at File > Preferences > Sketchbook location. By default, the sketchbook folder is located inside the portable folder when the Arduino IDE is running in portable mode. However, this makes it easy to lose the contents of the sketchbook folder when you update to a new version of the Arduino IDE, so I recommend setting the sketchbook folder to some other non-user specific location on the computer.

facchinm, 2019-01-29:
@aroedercls you can copy/paste the folder, and since you are on Windows 10 there's also no need to install the drivers manually on every PC.

aroedercls, 2019-01-29:
One last question, with Arduino 1.8.8 and using ArduBlock, the pin for creating a blink has been removed.  Has it been relocated or is it gone entirely?

[chrome_2018-06-11_10-49-25]
Ashley Roeder
Senior System Support Specialist

Creative Learning Systems
1801 Lefthand Circle, Unit A
Longmont, CO 80501
Preferred Phone: 800-458-2887
Direct Phone: 800-458-2880 ext. 132
aroeder@clsonline.com<mailto:aroeder@clsonline.com>
support@clsonline.com<mailto:support@clsonline.com> (Preferred)

closed 2020-08-02 by per1234