arduino/Arduino #8461: Library Manager, struggling to get examples to appear

WifWaf, 2019-01-29:
Hi there,

My library was recently uploaded: https://github.com/WifWaf/MH-Z19

This is almost certainly my error (I'm new to this). I'm struggling to get the examples to appear in Arduino IDE "FIle>Examples".

I originally uploaded the examples as .cpp, though changed the extensions/tested to .ino. The update has gone through, though they are still not appearing.  I'm using sub-folders which doesn't seem be an issue for other libraries.. The only thing I can think is that I have '&' sign as part of the name for one the examples - perhaps I'm wrong to assume the job would ignore this file only.

Are you able to provide any insight? Many thanks

PaulStoffregen, 2019-01-29:
Each example needs a folder with the exact same name as its main .ino file.

PaulStoffregen, 2019-01-29:
One additional comment - from a library design point of view, you can achieve much better compatibility with most boards if you use a begin() function which accepts a Stream reference.  Normally inside the library, you would use a Stream pointer, assigned by the begin(Stream &myport) function.  This way, users can pass a SoftwareSerial stream on boards like Uno, or a hardware serial stream on boards which have extra ports, or if strange cases like a special lib that does serial over some other media, if your library accepts a general Stream reference, you can automatically work with *anything* else that uses Arduino's Stream.h base class.  That's a much better way than restricting your code to only 2 boards and hard-coding a dependency on SoftwareSerial (when so many AVR & ARM boards do have extra real hardware serial).

closed 2019-01-29 by per1234