arduino/Arduino #8469: Please add my libraries to the Library Manager

bitbank2, 2019-01-31:
2 libraries to add:

https://github.com/bitbank2/Multi_OLED
https://github.com/bitbank2/Multi_BitBang

bitbank2, 2019-01-31:
Released both - thanks for the tip

bitbank2, 2019-02-01:
Anything else needed to get it added?

per1234, 2019-02-02:
Everything looks good. Now you only need to wait.

Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that.

cmaglie, 2019-02-05:
done!

closed 2019-02-05 by cmaglie