arduino/Arduino #8473: the arduino ide text not visible on linux

ijash, 2019-02-01:
[image]

here's my system configuration https://termbin.com/4k37

facchinm, 2019-02-01:
This looks a bit like https://github.com/arduino/Arduino/issues/7144  (also that one is on Linux Mint, so it could be an issue with the default theme). Probably something like https://github.com/arduino/Arduino/issues/7144#issuecomment-369415299 could fix the glitch.
@per1234 should we close it as duplicate and continue there?

skuroedov, 2019-02-15:
It is probably because of theme. I am using ArchLinux with XFCE and this happens when I am using some dark theme. In my case Arduino IDE is not the only program where it happens, it happens with html input fields too. Try to change your theme.
[image]
[image]
[image]

bjwest, 2019-07-11:
This problem was fixed in the Library Manager on 10 Sept, 2016 (issue [#5353](https://github.com/arduino/Arduino/issues/5353)) with commit [b17c512](https://github.com/arduino/Arduino/commit/b17c5122c2b44773ca75b87864c96a1e6b67d064).  A quick look though the code and I can't find a similarly named file for the Board Manager.  Anyone know where the style sheet for the board manager is and can set the color to black to see if that fixes this problem?