arduino/Arduino #8474: And another Library Manager and Board Manager issue with "package_index.json"

DSiwi, 2019-02-01:
On starting Library Manager and Board Manager, I immediately get the attached Error Log. 

 Error downloading https://downloads.arduino.cc/libraries/library_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading https://downloads.arduino.cc/libraries/library_index.json
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onUpdatePressed$2(LibraryManagerUI.java:206)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://downloads.arduino.cc/libraries/library_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:120)
	at cc.arduino.contributions.JsonDownloader.download(JsonDownloader.java:49)
	at cc.arduino.contributions.GZippedJsonDownloader.download(GZippedJsonDownloader.java:63)
	at cc.arduino.contributions.libraries.LibraryInstaller.updateIndex(LibraryInstaller.java:67)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onUpdatePressed$2(LibraryManagerUI.java:202)
	... 1 more
Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:192)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1946)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:316)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:310)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1639)
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:223)
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:1037)
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:965)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1064)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1367)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1395)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1379)
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:162)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:181)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 6 more
Caused by: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:397)
	at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:302)
	at sun.security.validator.Validator.validate(Validator.java:262)
	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:324)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:229)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:124)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1621)
	... 19 more
Caused by: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at sun.security.provider.certpath.SunCertPathBuilder.build(SunCertPathBuilder.java:141)
	at sun.security.provider.certpath.SunCertPathBuilder.engineBuild(SunCertPathBuilder.java:126)
	at java.security.cert.CertPathBuilder.build(CertPathBuilder.java:280)
	at sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:392)
	... 25 more

No libraries or boards are listed except the installed defaults.

I have tried all of the following links in Preferences/Additional Board URLs:
- http://arduino.esp8266.com/stable/package_esp8266com_index.json
- http://downloads.arduino.cc/packages/package_index.json
- https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

I have also:
- Deleted files and eventually the whole Arduino15 folder.
- Manually downloaded "package_esp8266com_index.json" and placed it in the Arduino15 folder.
- Tried on both ethernet and wifi.
- Inserted a firewall rule for Arduino.exe.
- IP v6 totally disabled in network configurations.
- Run "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true"
- Totally removed the IDE, deleted all associated files and folders, cleaned registry, searched for key phrases across my entire computer to ensure all files removed. Then reinstalled. Set the Pref-Manager URL and network proxy disabled. Still cannot load any library or board. 
- Run IDE at various times in protected mode.
- Installed the Hourly Build.

Note: I have a variety of environments (VS, Eclipse, JetBrains etc for various projects includig some Java development. 

Now at a loss for where to go next, especially since it works OK on my old laptop and not my main machine. Both are Windows 7 and similar install configurations. Any and all help appreciated.

per1234, 2019-02-01:
It looks like a duplicate of https://github.com/arduino/Arduino/issues/7948. Did you install the Arduino IDE via a command like:
```
flatpak install flathub cc.arduino.arduinoide
```

per1234, 2019-02-01:
Ah, I just noticed you're using Windows 7, so probably not.

DSiwi, 2019-02-04:
Correct - Windows 7, Although I have noticed that people have the same kind of issues whether Win or Linux etc.

facchinm, 2019-02-04:
@DSiwi can you reach  https://downloads.arduino.cc/libraries/library_index.json from a browser without any error? And which version of the IDE are you running? Since we bundle the Java jre we don't use system-wide certificate authorities; an old version of the IDE could fail to verify all SSL certificates released by authorities which were unknown at compialtion time (eg. Let's Encrypt)

DSiwi, 2019-02-05:
Yes, I can reach all three json’s mentioned (sorry, should have included that before).
This is the only version of the IDE installed. That is the latest version 1.8.8, installed using the windows install exe.

KebabLord, 2019-02-06:
I'm literally having same issue.

DSiwi, 2019-02-14:
I never did get the bone working properly and reading the json files. So as an experiment I got an old drive, installed a fresh copy of Win7 on it, set up Arduino etc again and lo and behold - works fine. So either it is something in the Windows system or in one of the many other compilers and IDEs screwing something up.
With the number of issues around this theme and the lack of reply means that the problem is probably never going to be resolved because, after all, it is only Windoze users anyway. The total lack of interest of Windows is evident with the lack of any effort to correctly integrate Arduino into the Windows environment. For example not even making a Start menu entry or using the very functional file selectors for opening or saving files.
Unfortunately this is an attitude that I have found very prevalent with Unix/Linux/BSD people. A total intolerance not only of Windows, but generally anyone who is at the NOOB end of the learning curve.

facchinm, 2019-02-14:
@DSiwi we provide a .exe that installs both the start menu integration + the .ino open on click, plus we also provide a Windows Store installer and a zip file for any other need, so I wouldn't say we are not Windows friendly :slightly_smiling_face: 
We tried very hard during the years to make the IDE installation as much "independent" from the host system as possible (by integrating the java jre for example) but some things are totally out of our control (installed antivirus, firewalls) and 99% of the times we are talking about Windows systems.

KebabLord, 2019-02-14:
I solved the problem now,
just tried to download the contents from another internet connection, my previous connection was a LAN (internet of the company I'm in). When I tried it with my home wifi it worked and downloaded the cleanly.

MattCosturos, 2019-05-03:
I found out that Arduino 1.8.9 installs it's own JRE with OUTDATED / MISSING CERTS

I had to go to https://downloads.arduino.cc/packages/package_index.json and then download all the certificates.
Then I had to manually import these certs into the cacerts file in the JRE which was installed in the Arduino directory! (not in the global java directory!)

Arduino installs it's own jre in C:\Program Files (x86)\Arduino\java\ and that cacerts file needs to be updated!

Kind of a shame that the currently shipping version has a jre with missing certs needed to download the board packages.

plasma-vision, 2019-05-22:
You can also try directing the jvm to use the Windows Certificate Store instead of the ancient cacerts file that comes with java 1.8. This may also help people behind a Cisco Umbrella filter.
One does this by adding the following lines to "C:\Program Files (x86)\Arduino\arduino.l4j.ini" (note, you will need to open a text editor as an administrator, then open the file in question, in order to edit it):
-Djavax.net.ssl.trustStore=NUL
-Djavax.net.ssl.trustStoreType=Windows-ROOT

sarah-g, 2019-06-24:
Also having the same issue with Arduino 1.8.9 on a Mac

jhuk, 2019-07-01:
plasma-vision's solution of editing arduino.l4j.ini solved the problem for me.
In my case I am using a work laptop that has Cisco Umbrella and would fail using any network. With the extra two lines it now works everywhere for me, thanks for the advise.

plasma-vision, 2019-07-01:
> Also having the same issue with Arduino 1.8.9 on a Mac
Unfortunately I don't have a ton of experience with Java and certificate management on the mac. I know the macos stores system level certs in the keychain, but I don't know how to make java look there. 
I'd recommend checking with your IT department to find out what certificates you need, and try importing them directly to the Java keystore.

a20781, 2019-07-11:
The same issue with Arduino 1.8.9 on a Mac(with java version 1.12) ， and have added three keystores in the MAC system cacerts, but still can not resolve this issue.

I am not sure whether Arduino 1.8.9 is using Mac system java? Or it is a bug for Arduino?

ximon, 2019-08-05:
Same issue for me over a corporate proxy, @plasma-vision 's work around gets it working for me too.

A6GibKm, 2019-10-29:
I am having the same problem when trying to build a flatpak for arduino 1.8.10 https://github.com/flathub/cc.arduino.arduinoide/pull/10

twling, 2020-01-31:
I want to add that I am experiencing the same errors as the original post, but I am using arduino ARM v1.8.11 on a Raspberry Pi 3 with the latest Rasbian. 

As noted in previous posts, the 'java/security/cacerts' file bundled is out of date, however it still contains the valid root certificate that is being presented from *.arduino.cc. 

I added debugging to the JAVA_OPTIONS (-Djavax.net.debug=ssl) and was able to trace that the actual error happening is that during the SSL handshake, the certificate chain is presented, but no match if found in the cacerts truststore. This is very unusual as a manual dump of the cacerts truststore shows an exact Root certificate match (COMODO ECC Certificatation Authority) down the the fingerprint, hashes, and digital signature.

I also tested out the beta version of Arduino 1.9.0, which does not have this problem and works perfectly (as far as downloading the library or board manager package list). Additionally, arduino ARM v1.8.10 also works fine. Even copying the cacerts file from either 1.9.0 or 1.8.10 into 1.8.11 does not correct the problem. To me this indicates there is something else wrong in v1.8.11 with how the certificate signatures are being compared.

For anyone else in the same situation of using Arduino IDE on Rasbian, my suggestion would be to skip v1.8.11 and either stay at v1.8.10 or try out the beta v1.9.0

GeNextAppLabs, 2020-02-02:
Thanks @MattCosturos and @twling for your valuable input and to @plasma-vision for your elegant solution, as I also had exactly the same issue, which is now fixed.

AndreasWaldherr, 2020-02-11:
I totaly agree with >GeNextAppLabs<   :-)
Had the same problem on raspian with IDE 1.8.11. The 1.8.11 HOURLY BUILDS from today also works. Thanks for the fix!!!

domenickp, 2020-03-12:
I had this issue, due to connecting through Cisco Umbrella.  @plasma-vision 's answer seemed spot on, but I'm on a mac.  On a mac, the Arduino app stores the parameters equivalent to the Windows .ini file in a plist:
`/Applications/Arduino.app/Contents/Info.plist`
Open that file with your favorite text editor, and find the section starting with:
```
<key>JVMOptions</key>
<array>
```
You'll see each parameter wrapped in `<string></string>`.  Add these two lines to the end of that section, which will tell the JVM to use the mac system keystore:
```
<string>-Djavax.net.ssl.trustStore=NUL</string>
<string>-Djavax.net.ssl.trustStoreType=KeychainStore</string>
```
Afterward, I was able to use the library manager after restarting the Arduino app.

plasma-vision, 2020-03-12:
@domenickp Thanks for answering this for the Mac users. I knew there had to be a way.

DSiwi, 2020-03-26:
After a year's absence on other things, I have come back to doing some embedded work again. Mainly because we are in Covid lockdown. 
I installed the latest version as of this date (1.8.12) on the same Windows computer as before, and lo and behold - no problems. So either it has been fixed (maybe the missing cert's mentioned above or something) or a Windows certificate problem (I have had issues with Firefox on some sites before). Anyway, ....... thanks.

closed 2020-03-26 by DSiwi

xeniluom, 2020-06-09:
Hi, i try version "install" and "zip" but, got the problem. If i try to go on board manager, i cant do nothing before crash with 
`2020-06-09T23:53:42.731Z INFO c.a.c.p.ContributionInstaller:314 [ContributionManager Update Thread] Downloaded package index URL=[https://downloads.arduino.cc/packages/package_index.json]
2020-06-09T23:53:42.731Z INFO c.a.c.p.ContributionInstaller:324 [ContributionManager Update Thread] Check unknown files. Additional package index folder files=[package_index.json], Additional package index url downloaded=[]
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base$9.onIndexesUpdated(Base.java:1400)
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:148)
        ... 1 more`

I try many thing, remove my bitdefender first, next i try to remove arduino15 and lunch again (if i try to lunch before deleting arduino15, arduino not lunch...) I try to put : 
`-Xms512M
-Xmx1024M
-Dfile.encoding=UTF8
-Djava.net.preferIPv4Stack=true
-Djavax.net.ssl.trustStore=NUL
-Djavax.net.ssl.trustStoreType=Windows-ROOT`

nothing work. Anyone can help me?

facchinm, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!