arduino/Arduino #8476: Ctrl y issued after destructive Ctrl z in verison  1.8.8 deletes text, causes auto save and Auto Compilation which generates Exception in thread "AWT-EventQueue-0" java.lang.InternalError: Text range not in document: 1975-1973

Candyrb, 2019-02-01:
While editing a Large sketch, and after many changes and intermittent saves, when trying to use a Ctrl z to correct a change the corrected text is all wrong and there are a number of deletions that occur that do not match the text before Ctrl z was issued. 
Next when trying to correct the change with a Ctrl y, an automatic compile happens which generates the following crash:
[Arduino IDE Crash - Feb 2019.txt](https://github.com/arduino/Arduino/files/2822219/Arduino.IDE.Crash.-.Feb.2019.txt)

Candyrb, 2019-02-04:
Martino,
Many thanks for the feedback.
I will test the updated IDE today and get back to you.

The one thing I have noticed is that the problem does not develop immediately but rather seems to occur when I have been doing a large number of edits.

The edits involve moving between function Tabs – I am currently using 13 at the moment, and adding code as well as updating the logic.

Then if I try to correct a change, especially using the Ctrl z function, that’s when it all goes pear shaped.

Candyrb, 2019-02-04:
Martino,
Seems there is still a problem with Ctrl z actions - Please see the attached dump.

My actions were as follows:
Added lots of code to one tab in my case this was to the 11 tab.
Made a save
Performed about 8 Ctrl z actions
The IDE crashed 

Regards,
Richard Candy
[Arduino IDE crash, 4 Feb 2019.txt](https://github.com/arduino/Arduino/files/2828112/Arduino.IDE.crash.4.Feb.2019.txt)

Candyrb, 2019-02-04:
> This looks like an rsyntaxtextarea <https://github.com/bobbylight/RSyntaxTextArea>  bug. Would you mind trying to reproduce it with latest Hourly build? Thanks!
>
>
> Sketch uses 39052 bytes (15%) of program storage space. Maximum is 253952 bytes.
> Global variables use 5455 bytes (66%) of dynamic memory, leaving 2737 bytes for local variables. Maximum is 8192 bytes.
> Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
> 	at org.fife.ui.rsyntaxtextarea.RSyntaxUtilities.getMatchingBracketPosition(RSyntaxUtilities.java:490)
> 	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.doBracketMatching(RSyntaxTextArea.java:776)
> 	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.fireCaretUpdate(RSyntaxTextArea.java:837)
> 	at javax.swing.text.JTextComponent$MutableCaretEvent.fire(JTextComponent.java:4394)
> 	at javax.swing.text.JTextComponent$MutableCaretEvent.stateChanged(JTextComponent.java:4416)
> 	at javax.swing.text.DefaultCaret.fireStateChanged(DefaultCaret.java:802)
> 	at javax.swing.text.DefaultCaret.changeCaretPosition(DefaultCaret.java:1274)
> 	at javax.swing.text.DefaultCaret.handleSetDot(DefaultCaret.java:1173)
> 	at javax.swing.text.DefaultCaret$DefaultFilterBypass.setDot(DefaultCaret.java:1913)
> 	at javax.swing.text.NavigationFilter.setDot(NavigationFilter.java:64)
> 	at org.fife.ui.rtextarea.ConfigurableCaret$FoldAwareNavigationFilter.setDot(ConfigurableCaret.java:724)
> 	at javax.swing.text.DefaultCaret.setDot(DefaultCaret.java:1151)
> 	at javax.swing.text.DefaultCaret.setDot(DefaultCaret.java:1051)
> 	at javax.swing.text.DefaultCaret$Handler.removeUpdate(DefaultCaret.java:1773)
> 	at javax.swing.text.AbstractDocument.fireRemoveUpdate(AbstractDocument.java:259)
> [... cut ...]
> 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
> 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
> 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
> 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
> 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Martino,
Seems there is still a problem with Ctrl z actions - Please see the attached dump.

My actions were as follows:

*	Added lots of code to one tab in my case this was to the 11 tab.
*	Made a save
*	Performed about 8 Ctrl z actions
*	The IDE crashed

Question:

Do you want to continue to use the Hourly build or go back to 1.8.8?

 

Regards,

Dr Richard Candy

 

Mobile: +27 82 803 3945
Email:  <mailto:richardcandy@mweb.co.za> richardcandy@mweb.co.za

facchinm, 2019-02-04:
I totally looks related to rsyntaxtextarea, I'd forward it to @bobbylight if it was a bit more reproducible :slightly_smiling_face: .
Using the nightly should not harm :wink: